SpacehuhnTech/esp8266_deauther #204: I dont know why it wont upload? its giving me this error, any help would be appreciated

marquette2019, 2017-04-20:
---Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
C:\Users\PRAVI_~1\AppData\Local\Temp\arduino_modified_sketch_24782\esp8266_deauther.ino:50:18: fatal error: data.h: No such file or directory

 #include "data.h"

                  ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

jLynx, 2017-04-20:
Try moving the files to a directory closer to the root
for example in your documents and then try again

marquette2019, 2017-04-20:
what files exactly?

marquette2019, 2017-04-20:
I followed this :http://www.instructables.com/id/DIY-Wifi-Jammer-With-ESP8266-and-Mobile-App/

jLynx, 2017-04-20:
The zip file you download from here, extract that to your documents and try again

marquette2019, 2017-04-20:
okay it did something different now: 
Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

sketch\APScan.cpp: In member function 'bool APScan::start()':

APScan.cpp:14: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

   results = WiFi.scanNetworks(false, settings.apScanHidden); // lets scanNetworks return hidden APs. (async = false & show_hidden = true)

                                                           ^

sketch\APScan.cpp:14:59: note: candidate is:

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\pravi_000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)

     int8_t scanNetworks(bool async = false);

            ^

C:\Users\pravi_000\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

marquette2019, 2017-04-20:
sorry for being so confused

jLynx, 2017-04-20:
@marquette2019  don't follow that tutorial. Only use the one on this GitHub page

marquette2019, 2017-04-20:
@jLynx looking at the one from GitHub, i havent done step 13 and idk how to i cant find it?

marquette2019, 2017-04-20:
@jLynx Its almost the exact same tutorial as well, its not working?

marquette2019, 2017-04-20:
I think I know whats wrong, it would be step 13 on the GitHub page, I just dont know how to get those two files?

marquette2019, 2017-04-21:
I found the files on GitHub, just how do I copy both? they arent in file format?

jLynx, 2017-04-21:
You will have all the files you need when you download it from GitHub

spacehuhn, 2017-04-21:
> Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

Read and follow the readme. We don't help you on errors you get from other tutorials (theone you linked is stolen and outdated).

closed 2017-04-21 by spacehuhn

RiccardoMax, 2017-04-21:
I solved this issue by saving the deauther on the desktop and then proceeding with the normal installation.