SpacehuhnTech/esp8266_deauther #207: 192.168.1.4 is not responsive.

ethanwit, 2017-04-21:
I've tried compiling with Arduino, and the flashing bin files method, same problem every time. 
I've tried this on multiple iphones, androids, and my computer using a wifi adapter. The wifi network is always visible, but I can only be connected and view the 192.168.1.4 page for about 15 seconds before it just starts infinitely loading. I have to forget the network and replug the device and still wait a bit before it let's me do it all over again. The thing is working, I was able to spam my network and my wifi analyzer showed hundreds of networks, so I know it works.
What do I do? Is my ESP defective? I have this https://www.amazon.com/gp/product/B010O1G1ES/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

RaphiSimonson, 2017-04-21:
It's 192.168.4.1, not the reverse.

jLynx, 2017-04-23:
What @RaphiSimonson said

closed 2017-04-23 by jLynx