SpacehuhnTech/esp8266_deauther #208: AP scan not working anymore

RaphiSimonson, 2017-04-21:
After my first compile last night through Arduino, everything was working perfectly.

Now, I can connect to the network and get to 192.168.4.1 and see the UI, but no APs are picked up. It all worked last night, and I changed nothing about the code/board. I know there have already been threads about this but I haven't been able to find a solution.

breogangf, 2017-04-21:
I am agree @RaphiSimonson, I compiled everything from master (commit e93b5dc) an hour ago, I uploaded the code to a fresh board and I am able to connect to the network and see the UI, but I am not able to see any APs.

`I am trying with Chrome app (57.0.2987.137) on iOS 10.3.1 `

jurgverhoeven, 2017-04-30:
I have the same problem.

spacehuhn, 2017-08-09:
Can someone still reproduce this error? I had a similar issue a few days ago but now it's working fine.

N0vaPixel, 2017-08-09:
Thanks for helping us !

Seems to be big issue. So it would be nice if you can intercept the raw response from the http request to '/APScanResults.json'.
And also get the debug output of your browser, in order to get possible errors.
To do that you may need to use a computer. On computer, browsers offer so much more tools for debugging web apps than on a mobile device.

Case 1: You get some errors and it may be easy for us to fix the issue.
Case 2: You didn't get errors on browser, and the feature to scan wifi networks works fine. This show that the html/js code is not fully compatible with some browser on mobile.

spacehuhn, 2017-08-09:
I had the problem that the ESP8266 crashed when scanning for APs on the web interface, it wasn't a problem on the OLED however. I was in a very dense WiFi area and I looked up if there could be an overflow or something, but it's working fine. Even if it finds the maximum number of 80 APs and all have a 32 char long SSID, no crash. 

Now that I'm home, I can't reproduce the error. Here are not enough WiFi networks for a good stress test.

N0vaPixel, 2017-08-09:
@spacehuhn To perfom a stress test. Why not use a second esp or use such tools like mdk3 to launch a beacon flood attack ?

spacehuhn, 2017-08-09:
@N0vaPixel I did that, it's not sufficient enough. You'd need multiple interfaces that send hundrets of packets on multiple channels to simulate a real enviroment.

tobozo, 2017-10-16:
@N0vaPixel @spacehuhn I'm currently testing a full SPIFFS implementation, whenever some JSON is produced, it is written to the flash, then sent via server.streamFile, so no size mismatch issues with this approach.

I'm still having the webserver crash in WiFi._scanDone when AP.results > 80. 
Actually the only way I found to avoid this crash is to disable the webserver, I'm not sure where to go from there.

tobozo, 2018-03-24:
fixed in the next version, closing this

closed 2018-03-24 by tobozo