SpacehuhnTech/esp8266_deauther #209: Error MSG

LiveChief, 2017-04-22:
Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
sketch\APScan.cpp: In member function 'bool APScan::start()':

APScan.cpp:14: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

   results = WiFi.scanNetworks(false, settings.apScanHidden); // lets scanNetworks return hidden APs. (async = false & show_hidden = true)

                                                           ^

sketch\APScan.cpp:14:59: note: candidate is:

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\prajw\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)

     int8_t scanNetworks(bool async = false);

            ^

C:\Users\prajw\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

LiveChief, 2017-04-23:
Fixed it :)

closed 2017-04-23 by LiveChief

jLynx, 2017-04-23:
What was the problem?

LiveChief, 2017-04-23:
Problem: Broken Board
Fix: Get a new board.

jLynx, 2017-04-23:
How did you know the board is broken?

LiveChief, 2017-04-23:
the lights never turned on