SpacehuhnTech/esp8266_deauther #210: Access Points tab refreshing constantly

enderface36, 2017-04-22:
Hey, I am have very basic knowledge so bare with me.
Issue: When i go to 192.168.4.1 the page is frozen other than the scan button flashing and the page constantly refreshing.
Hardware/software: I am using a NodeMCU w/ ESP8266, I used the arduino method of installation.
Plz help

jLynx, 2017-04-27:
Can you try clear your EPROM and reflash?

Mikubo, 2017-04-29:
How can i clear the EPROM?

jLynx, 2017-04-29:
@Mikubo read the wiki faq

Mikubo, 2017-04-30:
Thanks jLynx. I've read the wiki faq but there is no explanation as I have to use this.
Does this also works? http://s6z.de/cms/index.php/homeautomation-homecontrol/hardwareplattformen/esp8266/131-loeschen-des-gesamten-flashspeichers

closed 2017-08-09 by spacehuhn