SpacehuhnTech/esp8266_deauther #211: My deauth BUTTONS ARE GONE

Sumeetu47, 2017-04-22:
[image]
[image]
I am using lolin nodemsu 
i used auto-flash software provided on your page and it worked perfectly but after 6-7hrs the buttons were gone. it connect properly to every device, even scans ap s and clients and selects them perfectly. just does not show buttons to deauth or becon etc.. please help me

jLynx, 2017-04-22:
Can you try clear the EPROM and then reupload the sketch?

skeeeemer, 2017-04-23:
I had this issue too. I cleared the EPROM and then reuploaded it and now it works fine, cheers.

Sumeetu47, 2017-04-23:
Please Tell me How to

spacehuhn, 2017-04-23:
https://github.com/schinfo/EEPROM-Cleaner

Sumeetu47, 2017-04-23:
thanks for help ... there was no need to clear eeprom,..... i just reuploaded using your website page instructions.

closed 2017-04-23 by spacehuhn

Sumeetu47, 2017-04-23:
my deauth function is not working at all just says status 20pkts/s but nothing happens...