SpacehuhnTech/esp8266_deauther #214: Deauth with oled display enabled not working

gumslone, 2017-04-23:
Anyone experiencing the same issue as me?

When i use an oled display, and activate the deauth attack it has no effect on the wifi network,
so i have disable it in the code and then everything works as expected.

---

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

highground88, 2017-04-24:
I am finding the same - although I have tried flashing with NodeMCU flasher (works, no display activated in bin file) vs using Arduino IDE (doesn't work, display activated). It is showing packets being sent however. If I comment out the display, it works, however I also noticed that after the attack starts via the Arduino upload, connectivity to the device AP seems to drop on my phone - I don't recall a previous version doing this.

gumslone, 2017-04-24:
I have added additional menu point to send beacon attack, which works with oled display enabled no issues. Added a menu option to turn off display, because i thought that the display requires to much energy, even with turned off display the deauth attack is not working, though the serial monitor shows that the deauth attack packets are sent.
The code to turn display off is:
`    display.end();
    display.displayOff();
    pinMode(D1, OUTPUT);
    pinMode(D2, OUTPUT);
    digitalWrite(D1, LOW);
    digitalWrite(D2, LOW);`
which didn't help.

First i thought that my code modification has caused the problem, but later i tried the default code from this repo which didn't work for me too.

Only when i comment the #define USE_DISPLAY line, the deauth attack works.
Tried 3 different esp8266 development boards, even without the display connected, all of them show the same behavior.

tommasoconti98, 2017-04-24:
I also tried with the oled display, and to me it doesn't work the deauth attack.

a61412228, 2017-04-27:
I have the same problem, when I comment the line  #define USE_DISPLAY works perfectly, I will try with a greater power source.

spacehuhn, 2017-04-28:
Found the bug!
Will be fixed in the next update.

highground88, 2017-04-28:
Great work, thank you for your excellent project.

closed 2017-08-09 by spacehuhn