SpacehuhnTech/esp8266_deauther #215: NodeMCU Resets after selecting AP and then clicking the attack tab

rsegiun, 2017-04-23:
I'm having an issue where if you select an AP and then select the "Attack" tab the board resets.  I'm using a Lolin NodeMCU V3 which likely doesn't mean much since all of these boards are made in China and have different names.  I flashed the board as a 8266-23E with a flash size of 4M (1M SPIFFS).

---

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

rsegiun, 2017-04-23:
Here is the stacktrace:

Exception (28):
epc1=0x402108e2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff4ff0 end: 3fff5350 offset: 01a0

>>>stack>>>
3fff5190:  000008e3 00000001 3fff5250 4021173d  
3fff51a0:  3fff04e8 3fff51ec 3fff5250 3ffefd08  
3fff51b0:  3ffe87ac 3fff4258 3fff4258 40210962  
3fff51c0:  3ffe8640 0000003f 3fff5250 402109fe  
3fff51d0:  3ffe87ac 3fff4258 3fff5250 402078a8  
3fff51e0:  3fff95b0 0000002f 00000022 3fff9570  
3fff51f0:  0000002f 00000020 3fff5250 4021131a  
3fff5200:  3fff93d0 00000000 3fff5250 40211372  
3fff5210:  0000000a 3fff5300 3fff5250 402113a2  
3fff5220:  00000001 00000003 00000001 00000010  
3fff5230:  3fff4024 3fff3760 3fff5300 00000001  
3fff5240:  00000001 00000001 3fff7300 4020a75e  
3fff5250:  00000000 000008cf 000008c1 402113a2  
3fff5260:  4010684c 3fff7300 3fff5290 402104e2  
3fff5270:  0000000f 401011f3 3fff7300 4020d64a  
3fff5280:  4020d654 3fff7300 3fff52b0 4020d686  
3fff5290:  3fff9540 0000001f 00000010 402114c8  
3fff52a0:  4020d654 3fff7300 3fff3760 4020d841  
3fff52b0:  3fff8cd8 0000001f 00000010 4020c873  
3fff52c0:  3fff379c 401011f3 3fff4330 00000001  
3fff52d0:  00000000 0000000f 4020f610 3fff4330  
3fff52e0:  402012e6 00000001 00000001 3fff431c  
3fff52f0:  3fffdc20 000003e8 3fff3760 4020d93c  
3fff5300:  3ffe90b0 00000000 000003e8 0000f0bc  
3fff5310:  3fff3780 3fff93f8 00000016 40101941  
3fff5320:  4020f631 00000000 3fff4315 4020b95d  
3fff5330:  00000000 00000000 3fff4315 4020f659  
3fff5340:  00000000 00000000 3fff4330 40100114  
<<<stack<<<

rsegiun, 2017-04-23:
Exception (28):
epc1=0x402108e2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff4ff0 end: 3fff5350 offset: 01a0

spacehuhn, 2017-04-29:
Maybe an overflow. Should be fixed with the upcoming update.

closed 2017-08-09 by spacehuhn