SpacehuhnTech/esp8266_deauther #216: Wifi not working

tdbnz123, 2017-04-23:
HI There

I got a esp8266  board yesterday I flashed it with the bin file etc and was working good as a wifi jammer

I was playing around and (I could of touched a setting in the webpage of the wifi jammer )

now the wifi is not transmitting off the esp8266 board meaning I can't access the device so I can do the wifi jamming etc

I tried to reflash the board with software no luck seeing wifi network ?

is there any way I can reset board / or reflash with default out of the box software??

when I plug in to usb it blinks upon usb power

anyone got suggestion on solution on how to fix

cheers

jLynx, 2017-04-23:
Sounds like you made the SSID hidden. If you want to reset it just clear the EPROM

tdbnz123, 2017-04-23:
how do i clear Eprom ?
sorry I so new to these / adrenio board

cheers

jLynx, 2017-04-23:
The link to the FAQ would be a good place to start...smh

closed 2017-04-23 by jLynx