SpacehuhnTech/esp8266_deauther #218: Deauthing Networks that don't Support 802.11w but still won't Deauth.

CamGiles, 2017-04-24:
---
When I tried to deauth my home wifi network, the deauther did not work after several times of trying. I checked my router and it dosn't support the new 802.11w security patch. How should I fix this or what is wrong.

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

theblixguy, 2017-04-25:
Maybe your router is running on 5GHz, the ESP8266 only works with 2.4GHz.

CamGiles, 2017-04-25:
My router outputs a 5g signal and a 2.4 g signal

poslogic, 2017-04-25:
Ask your questions on your router's forum page

vbonk, 2017-04-25:
You might want to try Increasing the packet rate. See https://github.com/spacehuhn/esp8266_deauther/issues/104

theblixguy, 2017-04-26:
> My router outputs a 5g signal and a 2.4 g signal

Doesn't matter. If it's a 5GHz router then it wont work. You'll have to split it into two APs - one that accepts 2.4GHz devices and one that supports 5GHz devices. Most modern routers allow this option. I have a Smarthub router from BT, and I have the same issue, the deauth doesnt work on it no matter what, splitting into two APs is the only solution that I could find.

CamGiles, 2017-04-26:
I do have two APs, but still doesn't work.

CamGiles, 2017-04-26:
I am not having trouble connecting and finding my wifi. I can select it at the APs section. It is just when I try to press the deauth option under attack, it will say that it is attacking but it doesn't kick people off.

spacehuhn, 2017-04-29:
No issue here. It's a testing device and the attack is not working against your network, you should be glad.

closed 2017-04-29 by spacehuhn