SpacehuhnTech/esp8266_deauther #222: What can I do to fix this error (Last post got closed and I don't know why)

Endous, 2017-04-26:
Arduino: 1.8.2 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\owner\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\owner\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\owner\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10802 -build-path C:\Users\owner\AppData\Local\Temp\arduino_build_724702 -warnings=none -build-cache C:\Users\owner\AppData\Local\Temp\arduino_cache_57785 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6 -verbose C:\Users\owner\AppData\Local\Temp\Temp2_esp8266_deauther-v.1.3.zip\esp8266_deauther-v.1.3\esp8266_deauther\esp8266_deauther.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\owner\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\owner\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\owner\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,UploadTool=esptool,UploadSpeed=115200,FlashSize=4M3M -ide-version=10802 -build-path C:\Users\owner\AppData\Local\Temp\arduino_build_724702 -warnings=none -build-cache C:\Users\owner\AppData\Local\Temp\arduino_cache_57785 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6 -verbose C:\Users\owner\AppData\Local\Temp\Temp2_esp8266_deauther-v.1.3.zip\esp8266_deauther-v.1.3\esp8266_deauther\esp8266_deauther.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0
Using core 'esp8266' from platform in folder: C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0
Detecting libraries used...
"C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10802 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "C:\Users\owner\AppData\Local\Temp\arduino_build_724702\sketch\esp8266_deauther.ino.cpp" -o "nul"
"C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10802 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src" "C:\Users\owner\AppData\Local\Temp\arduino_build_724702\sketch\esp8266_deauther.ino.cpp" -o "nul"
"C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10802 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src" "C:\Users\owner\AppData\Local\Temp\arduino_build_724702\sketch\esp8266_deauther.ino.cpp" -o "nul"
"C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10802 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266mDNS" "C:\Users\owner\AppData\Local\Temp\arduino_build_724702\sketch\esp8266_deauther.ino.cpp" -o "nul"
"C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DARDUINO=10802 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer\src" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266mDNS" "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\EEPROM" "C:\Users\owner\AppData\Local\Temp\arduino_build_724702\sketch\esp8266_deauther.ino.cpp" -o "nul"
"C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++1 [... cut ...]
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WebServer 
Using library ESP8266mDNS in folder: C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266mDNS (legacy)
Using library EEPROM at version 1.0 in folder: C:\Users\owner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\EEPROM 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Endous, 2017-04-26:
If I can atleast get pointed in the right direction instead of my thread being put as 'Invalid' I can move on and use the NodeMCU but right now I keep receiving this error and no one seems to be helping.

tobozo, 2017-04-26:
Your previous issue probably got closed because pasting such a big compile log in an issue is the best way to piss of the project manager. Best practice with dense content is to paste your logs in pastebin or similar service and provide the link.

Anyway, from the logs it seems like you manually installed and misplaced the modified ESP8266Wif library provided with this project, and now it's conflicting with the one provided with the Arduino SDK 2.0.0. 

As per the ReadMe, you're supposed to *overwrite* the ESP8266Wifi library in the SDK2.0.0 library folder.

closed 2017-04-26 by jLynx