SpacehuhnTech/esp8266_deauther #224: every time i try  do this i get an error

pandasniper100, 2017-04-26:
---Sketch uses 547706 bytes (52%) of program storage space. Maximum is 1044464 bytes.
Global variables use 56276 bytes (68%) of dynamic memory, leaving 25644 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

closed 2017-04-26 by spacehuhn

pandasniper100, 2017-04-26:
why is it invalid ?

pandasniper100, 2017-04-26:
what am i doing wroung ?

pandasniper100, 2017-04-26:
well this was a waist of time

pandasniper100, 2017-04-26:
ok so fixed it my self used a diffrent cable thanks for the help cough cough

mcmasterp, 2017-04-26:
The reason you get no help is that your issue is not an issue with this project but your ability with arduino and uploading code. Google would be a faster way to fix those issues.

pandasniper100, 2017-04-26:
hay i read the full install you could of at least said oh have you tryed this or that to troubleshoot instead i got ignored and no it was nothing to do with my abilty with ardunino it had to do with the cable i was using to falsh iti was using one that charges my tonbux power bank but that did not wotk so i used the one for my phone charger witch did so yhea

jLynx, 2017-04-26:
Not sure if you realise but this is not a help and support section for your esp device. This is a section for posting bug reports for the code. Your "issue" had nothing to do with the code, so it had nothing to do with us