SpacehuhnTech/esp8266_deauther #228: Access Denied after attack

Voxzaco, 2017-04-27:
Android phone unable to connect to "pwned" after one attack. 
Error: "Denied Access to network. Connection failed." 

Anyone else?

Mikubo, 2017-04-27:
have you increased the packetrate to high? Increasing the packetrate to high and choosing to much ssids the  ESP8266 CH340G V3 stucks and you have to disconnect and connect it again. For me, firering a single SSID works also with an max. packetrate  of 250. You'll have to experiment with the packetrate and number of ssids you want to deauther.

jLynx, 2017-04-27:
Can you try setting your channel to the same as the router you are trying to deauth?

drunkinmky, 2017-04-28:
i need help everytime i try to upload i get this

Board nodemcu (platform esp8266, package esp8266) is unknown

Error compiling for board NodeMCU 0.9 (ESP-12 Module).

what does it mean and what do i have to do your help is greatly appreciated ( P.S. i am new to arduino )

poslogic, 2017-04-28:
If you are new to ESP8266 and Arduino then please try and upload beginner sketches, so that you understand how to communicate with your board first.

https://www.arduino.cc/en/Tutorial/HomePage

drunkinmky, 2017-04-28:
ok i did it and i uploaded the blink to my board but i still get this

Board nodemcu (platform esp8266, package esp8266) is unknown

Error compiling for board NodeMCU 0.9 (ESP-12 Module).

is there something i am doing wrong

closed 2017-04-28 by Voxzaco