SpacehuhnTech/esp8266_deauther #229: ESP communictation failed

guiume123, 2017-04-27:
I did everything it said and now this?
Please help me
---Arduino: 1.8.2 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Archiving built core (caching) in: /var/folders/tp/3fwr85px5rn988yxv8qgwr0m0000gn/T/arduino_cache_623206/core/core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadTool_esptool,UploadSpeed_115200,FlashSize_4M3M_a10e394a73e62bc40415ab971368d8dc.a
Sketch uses 547132 bytes (52%) of program storage space. Maximum is 1044464 bytes.
Global variables use 55726 bytes (68%) of dynamic memory, leaving 26194 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Read [issues.md](https://github.com/spacehuhn/esp8266_deauther/blob/master/.github/issues.md) or your issue might be closed and labeled as invalid.

---

jLynx, 2017-04-27:
What com port are you selecting, are you sure you have selected the correct device?

spacehuhn, 2017-04-29:
No issue. See tge FAQ.

closed 2017-04-29 by spacehuhn