SpacehuhnTech/esp8266_deauther #266: how i turn of the built in led ?

Giuseppe2018, 2017-08-08:


spacehuhn, 2017-08-08:
Go to settings and change `Use LED`

[image]

Giuseppe2018, 2017-08-08:
It stays on

BlueArduino20, 2017-08-08:
Yep, I'm using the 1.6 version and the LED always stays on.

spacehuhn, 2017-08-08:
Works perfect for me on every board.
Are you maybe talking about the Deauther boards? They have multiple LEDs for the charging circuit and a LED on GPIO 16: https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/esp8266_deauther.ino#L20

N0vaPixel, 2017-08-08:
WoW. Never seen a LED on the GPIO 16 of my board. I use a classic NodeMCU board.

For me this is a red LED and is turned on at the beginning of the firmware.

Giuseppe2018, 2017-08-08:
Im talking about nodemcu esp8266

N0vaPixel, 2017-08-08:
You are talking about this red LED ?
[image]

spacehuhn, 2017-08-08:
Thanks for the picture @N0vaPixel !
Seems to actually be GPIO16. Not all of my boards have a LED there which is weird.

I will adjust the .bin file.

Giuseppe2018, 2017-08-08:
@N0vaPixel Yes

N0vaPixel, 2017-08-08:
To clarrify for all, there is currently two differents LEDs used by the firmware.

The blue is the default used when an attack is running. (Pin 2). You can configure it in the settings web page.

But the red is turned on at startup (GPIO16 confirmed)
This is not bad, it can show that the esp8266 is on and the firmware works correctly ¯\_(ツ)_/¯

spacehuhn, 2017-08-08:
Yep.

I updated the standart 1mb .bin file:
https://github.com/spacehuhn/esp8266_deauther/releases/download/v.1.6/esp8266_deauther_1mb.bin

I guess most people don't want that LED to be on, just because previous versions didn't had that either.
If you still want it, flash the Arduino code.

I will add it to my to-do list for the next update to put an option for GPIO16 into the web interface as well!

BlueArduino20, 2017-08-08:
Thanks! I was about to modify it.

N0vaPixel, 2017-08-08:
@spacehuhn Yes i think it's a good idea. This new LED can be confusing for many.

For those who wants to build a custom firmware 🔨 
In esp8266_deauther.ino
Replace this line #define USE_LED16
With //#define USE_LED16

closed 2017-08-08 by N0vaPixel