SpacehuhnTech/esp8266_deauther #267: Bug with scanned stations if you re-scan for networks

samdenty99, 2017-08-08:
Latest version affected: yes

### How to replicate

1. Scan for WiFi networks
2. Select a network that your want to scan for stations
3. Go to the stations page and scan
4. Once the scanning is complete, go back to the WiFi networks page
5. Scan for new networks
6. If there are different networks, going back to the stations page will change the network displayed next to the Stations

spacehuhn, 2017-08-08:
Thanks!
I guess the station-list is not cleared when scanning for new APs. It saves the old ID from the AP, but by scanning fo new APs the IDs may change therefor making a faulty station list.

samdenty99, 2017-08-08:
Also when creating Beacon SSIDs, putting quotes in the Beacon SSID field will break the `onclick` handler. This can be fixed by replacing quotes with escaped ones:

    var resApsI = res.aps[i].replace("'", "\\'"); // Escape single quotes from network name
    resApsI = resApsI.replace("\"", "&quot;"); // Escape double quotes from network name

tobozo, 2018-03-24:
fixed in the next version, closing this

closed 2018-03-24 by tobozo