SpacehuhnTech/esp8266_deauther #269: Encrypt Wifi Clones

Mr-iX, 2017-08-08:
For me the feature to encrypt the cloned Network does not work. 
When I add SSIDs with the checkbox „Encrypted“ checked and start the Beacon Attack all Networks are still open/unencrpted. 

Does anybody else have this issue?

spacehuhn, 2017-08-08:
Works fine for me with v1.6. Both when cloning or adding it manually to the List.

Mr-iX, 2017-08-08:
Ok. I will try it tomorrow with a other NodeMCU.

MemesGeneral, 2017-08-08:
Another one

[image]

Mr-iX, 2017-08-09:
Now I have tested it with a other NodeMCU. When I search on my Arch Linux Machine, the networks are shown as encrypted, but on my iPhone they are all shown as unencrypted.

Can you reproduce this behavior?

[image]
[image]

spacehuhn, 2017-08-09:
Seems to be an IOS issue. Maybe something with the beacon frames?

Mr-iX, 2017-08-09:
I think so, because the other networks are show correctly. 
I don’t know how to debug that directly in iOS. Maybe it helps to sniff the packages with a WiFi chip in promiscuous mode.

Behooz-Sirouszad, 2017-09-24:
i have the same problem with ipad and iphone users they find easily which one is their correct AP

hmpowerbt, 2018-02-11:
I dont have the same problem. They are encrypted on IOS for me.

spacehuhn, 2018-03-24:
Please try if it works with the new 2.0 version

Mr-iX, 2018-03-24:
So now in the new version 2.0 it works!

Thanks for all the effort you put into this new version :smile:

spacehuhn, 2018-03-24:
Thanks! Great to hear that it works now 😃

closed 2018-04-04 by spacehuhn