SpacehuhnTech/esp8266_deauther #354: Important Information - WiFi deauther OLED V2

schinfo, 2017-08-09:
Dear Customer,
 
Thank you. I have already sent WiFi deauther OLED V2 to you.
But I fund an important problem.
When you receive the product.Please check the FS8205A on the board. Looks like the picture attached. It writes "CS078" on the chip.
If it is “CS078”. Please do not use it. I can refund you or send you a new one. Because this is a fake chip of FS8205A. It would case a damage heat to the board and may be risk for you.
I have already fixed it and changed it to other FS8205A. I do not know who of you will receive the risk board. So Please check it first.
If it is not then you can use it with no problem. 
Sorry for this, next time I will check it carefully.
 
Travis Lin
Make something amazing with you.
www.dong-sen.com

spacehuhn, 2017-08-09:
# If it says `CS078` like on the picture, then it's fake and you should contact the seller!

[image]
[image]

tobozo, 2017-08-10:
It's very nice for the seller to be transparent on this and it is very much appreciated, but isn't there a risk that someone will falsely claim their build is faulty just to get one extra deauther free ?

closed 2017-08-19 by spacehuhn