SpacehuhnTech/esp8266_deauther #355: anyone managed to successfully load it on an ESP32 board?

mbartov, 2017-08-10:
Bought this board: https://www.banggood.com/Wemos-Lolin-ESP32-OLED-Module-For-Arduino-ESP32-OLED-WiFi-Bluetooth-Dual-ESP-32-ESP-32S-ESP8266-p-1148119.html?rmmds=search

Still struggling to adapt the instructions and code for this board. Wondering if anyone else was luckier?

spacehuhn, 2017-08-10:
The code is not compatable with the ESP32 and will never be.
There is no way to inject packets on the 32 and even if someone finds a way to do that, you would have to rewrite the whole firmware.

mbartov, 2017-08-10:
Thanks Stefan for the clarification and saving me the frustration :-)

spacehuhn, 2017-08-10:
No problem. And don't worry, I have other projects coming up regarding the ESP32 😃

harshil2105, 2017-08-10:
Care to share what we would see in your projects related to esp32???

tobozo, 2017-09-21:
Just follow @spacehuhn on github and twitter and you'll know when something new got shared.

Closing this as it's more a discussion than an issue.

closed 2017-09-21 by tobozo

spacehuhn, 2021-12-12:
Hi! There are many issues on this repo about ESP32 support, so I published a blog post to bring some clarity to this topic: https://blog.spacehuhn.com/esp32-deauther/ 
TL;DR: It's made for the ESP8266, and I have no plans to port this application to the ESP32. But it's open-source, so you're free to do it yourself if you want. I appreciate your understanding.