SpacehuhnTech/esp8266_deauther #356: Deauth failure

harshil2105, 2017-08-10:
I can't deauth a essid...it gives me a lot of errors like attackinfo.json not found and other related errors and then i have to reload the page or restart the esp8266...I'm using a nodemcu and i haven't faced such issues in other releases of esp8266 deauth firmware

tobozo, 2017-08-10:
power issues can feature the same symptoms

harshil2105, 2017-08-10:
Elaborate power issue

tobozo, 2017-08-10:
one of those but list in non exhaustive : 
- battery is nearly empty
- the USB cable does not provide enough amps (powering from a mobile phone?)
- something else (i.e. custom hardware such as rgb OLED) is drawing power

harshil2105, 2017-08-10:
Thanks for the info...you may close this issue

closed 2017-08-10 by tobozo

spacehuhn, 2017-08-10:
Also keep in mind that when you start an attack on a network that the ESP maybe has to change the WiFi channel, thus changing the channel of its hostet `pwned` network too.
And changing the channel can result in connectivity issues, so you maybe have to reconnect.

harshil2105, 2017-08-11:
Yes i have faced this issue a lot....but haven't faced such issues in your
earlier releases