SpacehuhnTech/esp8266_deauther #358: Deauth attack not working

ThomSeagal, 2017-08-11:
When I do a Deauth attack I get the error error loading attackStart.json?num=0
How to fix it?

N0vaPixel, 2017-08-11:
This error can mean that the esp8266 is not responding after trying to start an attack.
So we need more informations. Can you provide the output of the serial communication with your esp ?

ThomSeagal, 2017-08-12:
Where can I find the the output of the serial communication?

andrew77768, 2017-08-12:
I want to add, I've got the same issue. It starts by saying the JSON attack file couldn't be loaded, and then shortly after I get the same error as OP. Like OP, I don't know where the output is, but if the thread is updated with instructions then I'll will send them over.

N0vaPixel, 2017-08-12:
Arduino IDE : Menu Tools --> Serial monitor.
Use it when connected to the esp.
You'll get additional debugging informations about what's happening during the firmware execution.

andrew77768, 2017-08-12:
Okay. Thanks for the promt reply. So, I selected the target network, went to attacks saw the correct network was in the selected APs so continued to start the Deauth attack, first it said there was an error loading the file, it continued and the status changed to starting. Next it created the attackStart.json?num=0 error. The monitor outputted this during that time:  https://pastebin.com/ek8PVYRp and the button to stop the attack wouldn't work as it was stuck on starting

spacehuhn, 2017-08-12:
That looks like the wrong baudrate. On the serial monitor in the bottom right corner, set it to 115200.

ThomSeagal, 2017-08-12:
with 115200 the output stays empty. But now I get this error in the arduino coding screen, java.io.IOException: jssc.SerialPortException: Port name - COM4; Method name - setEventsMask(); Exception type - Can't set mask.

http://puu.sh/x8nG4/0c4c6ea876.png

ThomSeagal, 2017-08-12:
I think I got it figured out, according to the serial monitor everything is working, but I still get the error. I ignored the error and pressed stop, it stopped, then I pressed start again and it worked without any errors. I will add a pastebin link down below.  It looks like it is working fine. Only problem is the wrong error I guess.

https://pastebin.com/JUDqFrAx

andrew77768, 2017-08-12:
Okay. https://pastebin.com/PpL1XPbD That's the log. It continued the spam after i pasted it but it was the same line "running Deauth attack... done". It was while I got that spam that I got the errors.

ihatecreatingaccounts, 2017-08-26:
nice theres no support here. this product is not working fuckers

ChiLegend, 2017-10-01:
I am also have this problem

ChiLegend, 2017-10-01:
But I actually can't deauther

ihatecreatingaccounts, 2017-10-01:
Its Not working. Its a scam

ChiLegend, 2017-10-01:
What do you mean scam?

ChiLegend, 2017-10-02:
I just switched to a lower version and It works lol

tobozo, 2018-03-24:
fixed in the next version, closing this

closed 2018-03-24 by tobozo