SpacehuhnTech/esp8266_deauther #362: pwned not coming on list.

farazappy, 2017-08-13:
Hello,
I am trying to upload this in ESP-01 board. 
It successfully uploads, but while the board is powered up my devices can't see any Wi-Fi networks or connect to previously saved networks. "pwned" also doesn't shows up in the list.

Help me please.

I am using 1mb version

twosdai, 2017-08-29:
Having the same issue

osysltd, 2017-08-29:
Reset the phone and try to connect from scratch to the re-flashed device.

andrew77768, 2017-09-04:
AI-THINKER is the default SSID, usually means you didn't flash it right. Try again and see if you see the new SSID

tobozo, 2017-09-04:
[reminder](): USE the sdk 2.0.0, IGNORE the message from the Arduino IDE telling you tu update to sdk 2.3.0

[image]

farazappy, 2017-09-04:
I was using it just above my router. Taking it away from the router (other room) works fine. What can be the problem here?

tobozo, 2017-09-04:
Do you mean interferences?
One way to know is by changing the channel used by your router (or by the deauther if you can access the Web UI).

For example assign channel 11 to the router and channel 1 to the deauther so the frequencies don't overlap.

If it solves the problem, then yes it's interferences.

tobozo, 2017-09-21:
Closing this as it is not related to a bug in the source of this project (more about antannae).
Feel free to reopen if I'm wrong.

closed 2017-09-21 by tobozo

7o79, 2017-09-26:
Hey I had everything working when I Accidentally pressed reset on the board and now every time that I upload the esp8266_deauther to it it just doesn’t show pwend  in my WiFi