SpacehuhnTech/esp8266_deauther #363: Oled usability screenshots preview for new users

mrhut10, 2017-08-14:
I've been looking all throughout the interwebs for a screenshot / menu layout of theoled screen.

I've ordered your version so no doubt will see it in person shortly but a layout of the menu would be really awesome.

tobozo, 2017-08-19:
It might be a bit off topic but you can "capture" an image from the OLED buffer (128*64 = 8 Kilobits = 1 Kilobyte) by accessing `display.buffer` at runtime.

spacehuhn, 2017-08-19:
And you can see it here:
https://www.tindie.com/products/lspoplove/wifi-deauther-oled-v15pre-flashed/
and here:
https://www.tindie.com/products/lspoplove/wifi-deauther-oled-v2-include-case-and-antenna/
and here:
https://github.com/spacehuhn/esp8266_deauther#adding-an-oled-display

dwillmore, 2017-09-10:
@mrhut10 are you looking for a description w/pictures of what the contents of the OLED screen show when using it?

tobozo, 2018-03-24:
closing this as solved

closed 2018-03-24 by tobozo