SpacehuhnTech/esp8266_deauther #364: Only some of the devices on the selected network are affected by the attack

sy9039, 2017-08-15:
Only some of my devices are being affected. I can attack the network and have some devices be disconnected but then use other devices on the same network and not be affected while the same attack is going on.

spacehuhn, 2017-08-15:
From the readme:
```The deauth attack will, if the connection is vulnerable, disconnect the devices from the network.```
There is no guarantee that it will work.

ExploiTR, 2017-08-15:
@spacehuhn  How much can I increase the deauth packaterate without making it unstable ?

spacehuhn, 2017-08-15:
depending on how many targets you have selected. If you target one network and broadcast, then I would say up to 100 is ok. You could also go up to 250, but that could already make the webserver unstable.

sy9039, 2017-08-15:
Is there any way that I could deauth all the device on the network

tobozo, 2017-09-21:
@sy9039 you can only deauth *vulnerable* devices on a given network, it may or may not be "all" devices.

If you think the WebServer is the bottleneck, there's a [serverless](https://github.com/spacehuhn/esp8266_deauther/tree/serverless) branch you can try to build, but you need to have dstike's OLED build, and it hasn't been fully tested.

closed 2017-09-21 by tobozo