SpacehuhnTech/esp8266_deauther #566: Limiting timeout value.

ExploiTR, 2017-08-17:
I've watched your [video](https://youtu.be/qcAcQQPRYrk) at YouTube and really got depressed thinking that some of this videos are getting more than 10^5 views. And, if people knows about this project in a large number, WiFi will became really useless technology.

In you'r projects latest release (1.6) I've saw that you made the timeout value can be set indeterminate. 

So, If you really wan't to make your project just only **deauther** , I would propose to limit the deauth timeout to 5 seconds and archiving(removing) all the previous releases.

This is really not an issue for your project, but hugely for the peoples who are innocent.

ExploiTR, 2017-08-17:
Though, there are other programmers like samdenty99 , who have also released this project without no deauth timeout limit.

Galenika, 2017-08-17:
Go and cry somewhere else

ExploiTR, 2017-08-17:
@Galenika  Are you the project author ?

ExploiTR, 2017-08-17:
I've opened the issue because the @spacehuhn _himself_ told that **he wants** the project only to be called as **DEAUTHER**

Galenika, 2017-08-17:
Ask yourself why your neighbours are JAMMING your wifi. Thats prob. why you are here.

ExploiTR, 2017-08-17:
Upgrade your devices. No one will be able to JAM your wifi. https://en.wikipedia.org/wiki/IEEE_802.11w-2009

Galenika, 2017-08-17:
Goodluck.

@spacehuhn Close this. And have you had thoughts of having a discord channel??

ExploiTR, 2017-08-17:
And @Galenika , if your `neighbours are JAMMING your wifi` , you should call the police. And, with some knowledge and a laptop with linux, you will be easily able to find who the evil guy is.

Galenika, 2017-08-17:
@exploitr You can JAM with linux, so are you gonna email all linux distrubitions? Check how many views there are on thoose youtube videos.

spacehuhn, 2017-08-17:
@ExploiTR the timeout already filters out enough people from using it to block a connection constantly. 
I don't want to punish people who are interested in the topic by restricting my program.

And @Galenika you don't behave very kind. You're not helping anyone here with your comments.
So please stop.

And for both of you, here is why it isn't jamming: https://www.youtube.com/watch?v=6m2vY2HXU60

closed 2017-08-17 by spacehuhn