SpacehuhnTech/esp8266_deauther #569: Code shows error exit status 1 no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

pks9862728888, 2017-08-19:
Please help me...your code shows compilation error:
[image]
exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

spacehuhn, 2017-08-19:
Follor step 12 to 14: https://github.com/spacehuhn/esp8266_deauther#compiling-the-source-with-arduino

Also look for similar issues next time before opening a new one!
https://github.com/spacehuhn/esp8266_deauther/search?q=ESP8266WiFiClass%3A%3AscanNetworks%28bool%2C+bool%26%29&type=Issues&utf8=%E2%9C%93

closed 2017-08-19 by spacehuhn

AssadAnabosi, 2017-08-26:
Wifi.scanNetworks() it'll work at least with me