SpacehuhnTech/esp8266_deauther #570: Can connect to esp8266

FietsenRijder, 2017-08-23:
I installed software version 1.6.0 on my esp-8266 with the flasher and I changed the ssid from 'pwned' to '            ' and now it's not showing up on my list anymore. It is powered. Can you please help me.

XAS-712, 2017-08-24:
Can you still connect to the network which is powered by you esp-8266?
(I am only a Chinese student. Maybe there are some grammar mistakes in this comment. If there are, please forgive. Thanks.)

spacehuhn, 2017-08-24:
In the Readme and in the FAQ is explained how to reset the settings, please follow these instructions.

FietsenRijder, 2017-08-28:
@XAS-712 No I can't see the network with my android, but I do see it with my computer as hidden network. So I tried to connect with the password, but it failed.

@SpaceHuhn: for some reason I can't reset my esp-8266 v3 via arduino. And I don't know how to do it via the first method. Is there any other way I can fix this problem? I read on the internet that some people can't connect to their router becouse they put a space in the SSID my SSID is 6 or 9 spaces. Is that the reason why I can't connect?
Edwin

pG5heTMrlbBv0mcH, 2017-08-29:
Try to change the SSID in the Row 96 in the Settings.cpp.
Then upload the firmware again.

spacehuhn, 2017-08-29:
> for some reason I can't reset my esp-8266 v3 via arduino
What does that mean?

Btw, please make sure to have the current version 1.6 falshed on it.

FietsenRijder, 2017-09-01:
Thanks I reset it and it works now :)

closed 2017-09-01 by spacehuhn