SpacehuhnTech/esp8266_deauther #571: Deauth attack not sending packets

vincekirkov, 2017-08-24:
Hi, 
I have an issue with the deauth. The serial console shows: 

running Deauth attack... done
running Deauth attack... done
getting attacks JSON (238)...done
running Deauth attack... done

However, the device(s) stay connected to the network. In wireshark I do not see deauth packets being sent. 

The GUI also doesn't show any packets being sent.

Attack | Status | Start/Stop
Deauth | 0pkts/s | STOP

Not sure how to troubleshoot further...

The device is NodeMCU 1.0 CP1202 with ESP8266MOD

tobozo, 2017-08-24:
Are you using the binary or did you setup environment, compile & upload ?

It sounds like it was compiled with the SDK 2.3.0 instead of 2.0.0.

Another reason could be that all your devices are safe (all of them support and are connected through 802.11n).

vincekirkov, 2017-08-24:
You were right. After using 2.0.0 all works great. Thank you, sir!

closed 2017-08-24 by tobozo