SpacehuhnTech/esp8266_deauther #574: Compiling error

SIlentPain2608, 2017-08-27:
Arduino: 1.8.3 (Windows Store 1.8.6.0) (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

C:\Users\gluis\AppData\Local\Temp\Rar$DIa0.864\esp8266_deauther\esp8266_deauther.ino:66:18: fatal error: data.h: No such file or directory

 #include "data.h"

                  ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

XAS-712, 2017-08-27:
Check that if the data.h is exist.

SIlentPain2608, 2017-08-28:
where should data.h be located?

pG5heTMrlbBv0mcH, 2017-08-28:
esp8266_deauther/data.h

SIlentPain2608, 2017-08-28:
yes it exists

closed 2017-08-28 by SIlentPain2608

VIJAYLRAJANDEKAR, 2017-10-16:
Using NodeMCU 1.0 (ESP-12E Module). Getting follwing error while compiling esp8266 deauther.

Arduino: 1.8.3 (Windows 7), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\hp\AppData\Local\Temp\Rar$DI87.472\esp8266_deauther\esp8266_deauther.ino:66:18: fatal error: data.h: No such file or directory

 #include "data.h"

                  ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Please advice.

VIJAYLRAJANDEKAR, 2017-10-16:
Compiling error #574

sean-the-random, 2017-10-16:
Im lit gunna make a youtube video about haw to install it and uninstall it
fr! Go to my youtube ill make it today lator on @youtube sean_the_random

VIJAYLRAJANDEKAR, 2017-10-16:
Oh Thanks