SpacehuhnTech/esp8266_deauther #575: "no matching function for call to 'ESP8266WiFIClass::scanNetworks(bool, bool&)'"

SIlentPain2608, 2017-08-28:
There is a screenshot link of a marked piece of code:
http://prntscr.com/gdt7jh

Arduino: 1.8.3 (Windows Store 1.8.6.0) (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:404:5: error: previous declaration of 'int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t)' with 'C++' linkage

 int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb);

     ^

In file included from sketch\Settings.h:12:0,

                 from sketch\APScan.h:9,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:451:62: error: conflicts with new declaration with 'C' linkage

 int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb);

                                                              ^

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:405:6: error: previous declaration of 'void wifi_unregister_send_pkt_freedom_cb()' with 'C++' linkage

 void wifi_unregister_send_pkt_freedom_cb(void);

      ^

In file included from sketch\Settings.h:12:0,

                 from sketch\APScan.h:9,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:452:46: error: conflicts with new declaration with 'C' linkage

 void wifi_unregister_send_pkt_freedom_cb(void);

                                              ^

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:406:5: error: previous declaration of 'int wifi_send_pkt_freedom(uint8*, int, bool)' with 'C++' linkage

 int wifi_send_pkt_freedom(uint8 *buf, int len, bool sys_seq);

     ^

In file included from sketch\Settings.h:12:0,

                 from sketch\APScan.h:9,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:453:60: error: conflicts with new declaration with 'C' linkage

 int wifi_send_pkt_freedom(uint8 *buf, int len, bool sys_seq);

                                                            ^

sketch\APScan.cpp: In member function 'bool APScan::start()':

APScan.cpp:18: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

   results = WiFi.scanNetworks(false, settings.apScanHidden); // lets scanNetworks return hidden APs. (async = false & show_hidden = true)

                                                           ^

sketch\APScan.cpp:18:59: note: candidate is:

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)

     int8_t scanNetworks(bool async = false);

            ^

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

closed 2017-08-28 by jLynx

SIlentPain2608, 2017-08-28:
why is it invalid?

jLynx, 2017-08-28:
Follow the instruction correctly and you wont have this issue

pG5heTMrlbBv0mcH, 2017-08-28:
Did you use the version 2.0.0 of esp8266-sdk?
Try to follow this:[https://github.com/spacehuhn/esp8266_deauther#compiling-the-source-with-arduino](url)

ItsDank, 2018-03-08:
Im really not sure, but the wifi jammer gave me the same fucking issue when uploading, some one help us.

ItsDank, 2018-03-08:
and i know that i followed the instructions perfectly.. sooo idk

tobozo, 2018-03-08:
because "incapable of following instructions" is more often "too lazy to read the instructions"

jLynx, 2018-03-08:
If people are too lazy to read instructions or don't bother to search for past issues that are exactly that same issue they have and has already been answered in full by us previously then why should we bother wasting our time trying to help them out when they are already a lost cause?

ItsDank, 2018-03-08:
I went thru the instructions aswell as i could man, no need to get salted over a thread.

exploitagency, 2018-03-09:
Just flash the bin file from the release page.

Use a tool like whats available here:

https://nodemcu.readthedocs.io/en/master/en/flash/#nodemcu-flasher

esptool works great on Linux but I think Windows users usually use the NodeMCUFlasher.

Your welcome...

deantonious, 2018-03-09:
Gratulojn! Vi estas la unua, kiu ricevas respondon de EIFs! Se la instrukcioj ne funkcias, ekzistas nur du ebloj: vi ne legis la instrukciojn / via tabulo estas fiksa. Kaj bonvole, ne nomu ĝin martelo!

ItsDank, 2018-03-09:
The line that we were having a problem with is: results = WiFi.scanNetworks);, settings.apScanHidden); // lets scanNetworks return hidden APs. (async = false & show_hidden = true)

is there anyway some one could correct this? i'm not sure exactly what i'm doing anyways...

ItsDank, 2018-03-09:
Just being honest

cjohnweb, 2020-05-01:
I have a similar issue in my own project. I have this line

```
// Broadcast AP
WiFi.softAP(serverSSID,serverPass);
```

which is throwing error

```
error: no matching function for call to 'ESP8266WiFiClass::softAP(String&, String&)'
```

I feel like I must have setup my IDE wrong because this used to work.