SpacehuhnTech/esp8266_deauther #576: I don't know hot to call this problem...

SIlentPain2608, 2017-08-28:
There is a screenshot link of a marked piece of code:
http://prntscr.com/gdt7jh

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:404:5: error: previous declaration of 'int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t)' with 'C++' linkage

 int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb);

     ^

In file included from sketch\Settings.h:12:0,

                 from sketch\APScan.h:9,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:451:62: error: conflicts with new declaration with 'C' linkage

 int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb);

                                                              ^

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:405:6: error: previous declaration of 'void wifi_unregister_send_pkt_freedom_cb()' with 'C++' linkage

 void wifi_unregister_send_pkt_freedom_cb(void);

      ^

In file included from sketch\Settings.h:12:0,

                 from sketch\APScan.h:9,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:452:46: error: conflicts with new declaration with 'C' linkage

 void wifi_unregister_send_pkt_freedom_cb(void);

                                              ^

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:406:5: error: previous declaration of 'int wifi_send_pkt_freedom(uint8*, int, bool)' with 'C++' linkage

 int wifi_send_pkt_freedom(uint8 *buf, int len, bool sys_seq);

     ^

In file included from sketch\Settings.h:12:0,

                 from sketch\APScan.h:9,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/user_interface.h:453:60: error: conflicts with new declaration with 'C' linkage

 int wifi_send_pkt_freedom(uint8 *buf, int len, bool sys_seq);

                                                            ^

sketch\APScan.cpp: In member function 'bool APScan::start()':

APScan.cpp:18: error: no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

   results = WiFi.scanNetworks(false, settings.apScanHidden); // lets scanNetworks return hidden APs. (async = false & show_hidden = true)

                                                           ^

sketch\APScan.cpp:18:59: note: candidate is:

In file included from sketch\APScan.h:6:0,

                 from sketch\APScan.cpp:1:

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note: int8_t ESP8266WiFiClass::scanNetworks(bool)

     int8_t scanNetworks(bool async = false);

            ^

C:\Users\gluis\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:243:12: note:   candidate expects 1 argument, 2 provided

exit status 1
no matching function for call to 'ESP8266WiFiClass::scanNetworks(bool, bool&)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

jLynx, 2017-08-28:
Dont spam issues or you will be banned

closed 2017-08-28 by jLynx

SIlentPain2608, 2017-08-28:
sorry, but I have no clue how to solve this problem... I did anything like in the instruction videos but it doesn't work! Can you please tell me what I did wrong? It's really frustrating!

jLynx, 2017-08-28:
No idea what video you are talking about, but like I said follow the instructions here like we have already said and you won't have any issues. This is the last time I am telling you this. If you can't follow simple instructions then I cannot help you

bocha777, 2018-02-06:
a quien te comiste salame no ves que siguio los pasos y no puede si sabes la solucion y no queres contestar no contestes directamente

XAS-712, 2018-02-07:
You didn't patch some important files.