SpacehuhnTech/esp8266_deauther #577: Beacon Clone The Only Thing Working

cooledwhip, 2017-08-28:
I get no SSID for 2 of the features and the one feature that actually deauths the wifi doesn't even work.
http://imgur.com/a/uz8sZ

spacehuhn, 2017-08-29:
You have to add SSIDs to the list in order to send SSIDs out.

Also please update your device, you seem to use an old outdated version. Follow the instruction on this GitHub repository and not other tutorials in the webs, they're mosty 💩

cooledwhip, 2017-08-30:
Ok thanks! I'm not so good with this stuff so I will update you!

cooledwhip, 2017-08-30:
Should I be looking at how to flash it? or what tutorial specifically?

tobozo, 2017-08-30:
[From the ReadME](https://github.com/spacehuhn/esp8266_deauther#installation)

cooledwhip, 2017-09-02:
@spacehuhn  Awesome! Got it to work! On the flasher I thought the (Flash F) thing was going to flash it to my F drive on my PC haha... that's why I didn't do that. Anyway I got it working. The problem is I'm only getting like 2 stations that show up. One of the 4 cell phones in my house and my raspberry pi that is on... My megaPC doesn't show up, nor do other family member phones. Also it takes forever to find the stations. Any tips??

And the deauthorizer doesn't stay active for so long. It's only deauthorizing for a few minutes max. Thanks!!

And great project. All these criticisms aren't bashes towards you but are rather just things that I am experiencing. Your work is so genius and way beyond me. Thanks

FietsenRijder, 2017-09-02:
@cooledwhip 
The problem is I'm only getting like 2 stations that show up. One of the 4 cell phones in my house and my raspberry pi that is on... My megaPC doesn't show up, nor do other family member phones. Also it takes forever to find the stations. Any tips??
I have the same problem have you fixed it yet?

tobozo, 2017-09-21:
Closing this issue as it seems to be solved

@FietsenRijder not all devices are vulnerable or visible (ESP8266 can't see 5GHz, only 2.4GHz)
Also interferences can affect the ESP, don't put it over your router :-)

closed 2017-09-21 by tobozo