SpacehuhnTech/esp8266_deauther #578: Can't connect to "192.168.4.1"

SIlentPain2608, 2017-08-29:
Basically, it works! But I can't connect to "192.168.4.1" It works with my computer and the phone of my girlfriend but not with my phone!

pG5heTMrlbBv0mcH, 2017-08-29:
Did you get an IP address?

SIlentPain2608, 2017-08-29:
What do you mean?

XAS-712, 2017-08-29:
pG5 thinks your phone didn't get a valid IP address. So your phone also didn't know how to access 192.168.4.1.

SIlentPain2608, 2017-08-29:
How do I fix it?

twosdai, 2017-08-29:
Having the same issue as XAS-712

osysltd, 2017-08-29:
Reset the phone and try to connect from scratch to the re-flashed device.

XAS-712, 2017-08-29:
@twosdai Sorry, I can't understand it what you said. Could you point it out?

FietsenRijder, 2017-09-03:
@XAS-712 Your English isn't bad but twosdai has the same issue as yours.

herdir, 2017-09-14:
I do have a similar problem here.
I've a WiFi Deauther V1.5 (Pocket 8266), i can see the wifi with my laptop and phone.
I can connect to it, and i can ping it without packet lost.
But when i try to go to my browser and connect to 192.168.4.1 then it fails all the time (This site can’t be reached).
Seems like the webserver isn't up (the port 80 is well open when scanning)
I tried to reset the device but still the same.

tobozo, 2017-09-21:
@herdir using the deauther while the battery is charging may produce this
What does the serial output say ?

olmstead819, 2017-09-30:
Hi, i'm having the same problem here. The nodemcu connected to my pc for power is working but out of the home is not working... Help please?

twosdai, 2017-10-01:
Sorry everyone for how long it took me to comment back, basically I was just bad at imaging my device and didnt image it correctly the first time. Re imaging it worked for me and make sure you have the image pointing at the correct area.

TheVados, 2017-10-04:
> How did you solve this problem? My phone also connects to 192.168.4.1, but the other devices are connected.

colognaut, 2017-11-16:
same here. brandnew v2.5 board.

MrAppAndCrap, 2017-11-16:
Hey guys I had the same problem. You need to flash firmware first. 
I followed this tutorial and it worked fine after that. 
https://www.youtube.com/watch?v=r5aoV5AolNo

closed 2018-03-06 by spacehuhn

fabiom91, 2019-09-19:
So what's the solution??

twosdai, 2019-09-20:
Its been a while since I solved this problem I ended up just flashing the firmware again. Via what the tutorial told me to do.

fabiom91, 2019-09-21:
Thank you twosdai, I wasn’t able to flash my esp8266 (I only found instructions for windows) but I solved it just changing my WiFi password

Ritesh8732, 2020-12-07:
> I do have a similar problem here.
> I've a WiFi Deauther V1.5 (Pocket 8266), i can see the wifi with my laptop and phone.
> I can connect to it, and i can ping it without packet lost.
> But when i try to [...]

You figured out? 
I am also facing same issue.

Ritesh8732, 2020-12-07:
> I do have a similar problem here.
> I've a WiFi Deauther V1.5 (Pocket 8266), i can see the wifi with my laptop and phone.
> I can connect to it, and i can ping it without packet lost.
> But when i try to [...]

have you figured out ? i have the same issue don't know what to do

prostealth, 2022-01-20:
hi, this is happened to me today. i just updated my phone and when im trying to use my deauther i cant access the web UI. and i realized when typing on address bar showing search suggestion while it supposed to not show the suggestion because the network doesnt have internet connection. but it does. so thats mean youre  connected to wifi but the phone still can access internet using mobile network.
here is how to fix this. go to wifi setting and open wi-fi assistant. enable "stay connected" and set it to manual.

 [image]

JJDS, 2022-09-10:
in my case, on a Samsung S7, I solve the problem, using Chrome and, on the page :192.168.4.1,  browser settings (top right ) check Desktop site, and  ...voila access OK. Hope this help some of you.

JJDS, 2022-09-10:
following my previous post, I must say that I do not have the esp8226_deauther, so I do not try on it but on one of my onw apk using the esp32 in AP mode .I think that it will work also with all projects due to the fact that  the problem is smartphone 
settings

JJDS, 2022-09-11:
may be necessary to restart the smartphone

twosdai, 2022-09-12:
Can we lock this thread, I don't think it needs to be repeatedly Necro'd over and over again with slight variations around the same issue.

JJDS, 2022-09-12:
I do agree.
I have only sent more posts as soon as I discover them.