SpacehuhnTech/esp8266_deauther #579: error loading APScan.json

cyberman54, 2017-08-30:
After scanning APs by pressing the scan button in web UI, no AP list is displayed, and red "error loading APScan.json" pops up. Page reload after this results in empty page, chrome reports ERR_EMPTY_RESPONSE.

Android 7.1.2, Chrome 60.0.3112.107

tobozo, 2017-08-30:
Sounds like you're in a very busy area. Can you give more details about your hardware, binary/source version, flashing mode ?

spacehuhn, 2017-09-02:
What version are you using and can you try out if other versions have the same problem?

cyberman54, 2017-09-02:
i bought the oled v2 pcb from tindie. it came preflashed with version 1.6.

Will try to connect it to Arduino IDE for further debugging.

Tried it with two different smartphones / android versions. Problem persists.

Maybe hardware failure?

cyberman54, 2017-09-02:
not a too busy area, ~15 WLANs to scan.

cyberman54, 2017-09-02:
pcb seems to reset sometimes during operation when controlled by smartphone. This does not occur when controlled by hardware buttons.

spacehuhn, 2017-09-02:
Ok then you have the exact same error I had a while ago. But I can't reproduce the error anymore which makes it hard to fix it.

> ~15 WLANs to scan.

That's interesting... so it doesn't have to do with number of SSIDs.
Could it have to do with sepcial characters in the SSID or something like this?

It must be in the sendResults function https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/APScan.cpp#L139 because it's working over serial and the OLED. So it probably crashes when sending/generating the JSON response.

cyberman54, 2017-09-02:
I saw same error at other location in different environment, so i don't assume it depends on SSIDs.

cyberman54, 2017-09-02:
I connected the board to Ardunio IDE and logged with serial monitor. See result below. There is an exception after AP scan. How can i support you debugging? Please let me know what information or data could be useful for debugging, i will then post it here.

[deauther-debug1.txt](https://github.com/spacehuhn/esp8266_deauther/files/1272097/deauther-debug1.txt)

cyberman54, 2017-09-02:
If using hardware buttons to scan, the scan completes correctly.
>Scan results: 35
>[...]
>scan done
>sorting APs 
>[...]

cyberman54, 2017-09-02:
After some testing i don't think the problem comes from the sendsResults function. It looks like there is a general problem with the board's webserver and/or wifi communication to the smartphone. The webpages sometimes do not load completely. That means there is missing data in HTML code. How can this be further debugged?

cyberman54, 2017-09-02:
[Off topic: Tried to flash the oled v2 D-Duino board from tindie with Arduino IDE. It seems the board cannot be switched in flashing mode?

esptool.exe -vv -cd ck -cb 256000 -cp COM3 -ca 0x00000 -cf 
C:\.../esp8266_deauther.ino.bin 

esptool v0.4.6 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 256000
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat C:\.../esp8266_deauther.ino.bin success
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	serialport_receive_C0: 98 instead of C0
]

cyberman54, 2017-09-02:
[Off topic: I compiled the v1.6 code from here and reflashed the oled v2 module. Result is that it does not boot any more :-( bricked it? code compiled without errors, flashing was succesful.]

cyberman54, 2017-09-02:
Habe mal einen Strack Trace gemacht, ergibt folgendes:

Decoding 12 results
0x40288b70: chip_v6_unset_chanfreq at ?? line ?
0x4020e39c: ESP8266WiFiClass::_scanDone(void*, int) at C:\...\ESP8266WiFi\src/ESP8266WiFi.cpp line 952
0x40232fdb: scan_cancel at ?? line ?
0x40232fe6: scan_cancel at ?? line ?
0x40232dcf: scan_pm_channel_op_cb at ?? line ?
0x40233260: scan_clear_channles at ?? line ?
0x40223e65: pp_tx_idle_timeout at ?? line ?
0x4021d7ad: ets_timer_handler_isr at ?? line ?
0x4021d7f2: ets_timer_init at ?? line ?

Die Zeile 592 in der ESP8266Wifi.cpp enthält die Klasse:
void ESP8266WiFiClass::_scanDone(void* result, int status)

-> Irgendwo da scheint der Fehler ausgelöst zu werden?

spacehuhn, 2017-09-02:
@tobozo @N0vaPixel you guys have an idea?
I think I'll go and restructure the whole webserver stuff (when I get some time to do that...). But a quick and dirty bugfix would be much appreciated :D

tobozo, 2017-09-02:
I get the same error when running a wifiscan while already running a wifiscan.
Maybe startAPScan() needs more logic to avoid this situation ?

spacehuhn, 2017-09-02:
hm.. @cyberman54 when you scan with the OLED and then open the APScan page, does it crash or show the list without problems?

tobozo, 2017-09-02:
Bug confirmed and fix suggestion:

```
void startAPScan() {

  if(scanMode == "scanning...") return;
  (...)
}
```

and the error is gone

cyberman54, 2017-09-02:
@spacehuhn:
>when you scan with the OLED and then open the APScan page, does it crash or show the list without >problems?

Scanning with hw button and OLED -> ok
trying to open 192.168.4.1 -> ERR_EMPTY_RESPONE
trying to reload 192.168.4.1/apscan.html -> board resets

cyberman54, 2017-09-02:
[Off topic: i tried to compile and flash v1.6 with the mod suggested by tobozo. Arduino IDE compiles and flashes with no error. But Oledv2 board does not boot. When pressing reset button on board serial monitor shows with 74880 baud:

ets_main.c 
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

What's going wrong here?
]

tobozo, 2017-09-02:
@cyberman54 what settings are you using in the IDE ?

My settings:

Board: Generic ESP8266
Flash Mode: DIO
Flash Frequency: 40Mhz
CPU Frequency : 80Mhz
flash size: 1M (256K SPIFFS  <<< not sure about this one but it works
Reset Method: nodemcu
Upload speed: 921600

In the sketch I've commented out the SSD1306 init line, the build I have i is using the SH1106.
I'm flashing from the test branch.

cyberman54, 2017-09-02:
@tobozo flashing & booting not works with your settings, thanks!

I edited code like below, but still same behavior: ERR_EMPTY_RESPONSE after trying to scan via web UI.
I feel this looks like an issue with the webserver?

//==========AP-Scan==========
void startAPScan() {
  
  if(scanMode == "scanning...") return;
  scanMode = "scanning...";

tobozo, 2017-09-02:
@cyberman54 the fix is for a situation that was only triggered by the buttons when starting a scan while a scan was already running. Only the stack trace is the same as the one you get when running apScan from the web UI.

However, I have this ERR_EMPTY_RESPONSE & restart behavior too.
When I change the value of `maxAPScanResults` down to 40 in `APScan.h` the problem seems to go away.

`#define maxAPScanResults 40`

Does it solve it for your too ?

cyberman54, 2017-09-03:
The change makes it slightly more stable, but i still have problems loading the websites on the different tabs. Sometime pages are not fully loaded, sometimes ERR_EMPTY_RESPONSE.

tobozo, 2017-09-03:
yep that's only a quick&dirty mitigation, but I don't like it because it trims the AP list before the results are sorted.
Long story short if you're in an area where maxAPScanResults > results it's fine to lower this value.

As for the erratic browser behaviour towards HTTP connexions, you may find Firefox is accidentally better with the web UI, but it's only because it lacks a "reload" functionality.

My dummy analysis:
Most ERR_EMPTY_RESPONSE problems happen after the deauther switches modes between STA and AP. 
If your're near your home network, your phone may automatically reconnect to your home WiFi network while the deauther is scanning/attacking, meanwhile the JavaScript is still running and sending requests to 192.168.4.1, which will eventually end up in timeouts (8 seconds for JSON, 1mn for HTML). Nasty bonus if you have a captive portal as your browser may cache some of the responses.

Also I'm quite sure this version of ESP8266WebServer has random issues with some WiFi configs, especially when it changes or is reloaded all the time.

@spacehuhn async websocket + bundling the web UI as a progressive web app would solve many problems here
[And for better debug I found out this would help](https://github.com/spacehuhn/esp8266_deauther/commit/9d4a8621e64e491547bf148969d535576c4d1a59)
- added handlers for xmlhttp.onabort and xmlhttp.onerror to getResults() in functions.js
- added try/catch foor JSON.parse in apscan.js and attack.js

cyberman54, 2017-09-03:
@tobozo i tested with a "blank" smartphone which only knew the board's wifi network, to avoid wifi and/or mobile network switching problems.

spacehuhn, 2017-09-03:
@tobozo working on a project that uses ansync webserver and socket. But it doesn't run on the old sdk... :/

Rexxar32, 2017-09-13:
Hi, thanks for this project!!
Yesterday at my home I flashed a WeMOS D1 mini 4MB  and everythink work flawless (about 6 SSID or less arround).
Today at work i got "error loading APScan.json" (the amount of essid arround goes up to 80-90) and i think the esp8266 webserver crashes after that.

I got the ESSID number with "iwlist wlp4s0 scan | grep ESSID -c" if it helps.

Is it posible to add the "maxAPScanResults" in the settings options tab?
Regards!

tobozo, 2017-09-13:
@Rexxar32 The value of `maxAPScanResults` affects the amount of allocated memory, changing this requires to recompile the sketch to a different mapping.

While it may be possible to add a setting that overrides this value (must be smaller or equal to `maxAPScanResults`), it will not be satisfying as it will trim the list before it's sorted by RSSI.
Because of this _workaround_ you may have to run APScan multiple times before you see your target SSID appear in the results.

tobozo, 2018-03-24:
fixed in the next version, closing this

closed 2018-03-24 by tobozo