SpacehuhnTech/esp8266_deauther #581: Cannot connect to wifi on Linux

Mr-iX, 2017-08-31:
I have the bug, that I cannot connect to the wifi of my ESP-Deauther with a Linux maschine. I have tried several versions of Linux (Ubuntu, Arch Linux) on different Laptops.
On iOS/Windows the connection works fine. The deauther runs on the latest version 1.6.

Log of the authentication process:
`Aug 31 20:36:03 Arch NetworkManager[333]: <info>  [1504204563.3023] device (wlp3s0): supplicant interface state: disconnected -> scanning
Aug 31 20:36:03 Arch NetworkManager[333]: <info>  [1504204563.2054] device (wlp3s0): supplicant interface state: associated -> disconnected
Aug 31 20:36:03 Arch NetworkManager[333]: <warn>  [1504204563.1977] sup-iface[0x55e141814520,wlp3s0]: connection disconnected (reason -3)
Aug 31 20:36:03 Arch wpa_supplicant[423]: wlp3s0: CTRL-EVENT-DISCONNECTED bssid=[MACofESP] reason=3 locally_generated=1
Aug 31 20:36:03 Arch kernel: wlp3s0: deauthenticating from [MACofESP] by local choice (Reason: 3=DEAUTH_LEAVING)
Aug 31 20:36:03 Arch wpa_supplicant[423]: wlp3s0: Authentication with [MACofESP] timed out.
Aug 31 20:35:53 Arch NetworkManager[333]: <info>  [1504204553.1829] device (wlp3s0): supplicant interface state: associating -> associated
Aug 31 20:35:53 Arch kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlp3s0: link becomes ready
Aug 31 20:35:53 Arch kernel: wlp3s0: associated
Aug 31 20:35:53 Arch wpa_supplicant[423]: wlp3s0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Aug 31 20:35:53 Arch wpa_supplicant[423]: wlp3s0: Associated with [MACofESP]
Aug 31 20:35:53 Arch kernel: wlp3s0: RX AssocResp from [MACofESP] (capab=0x11 status=0 aid=1)
Aug 31 20:35:53 Arch NetworkManager[333]: <info>  [1504204553.1493] device (wlp3s0): supplicant interface state: scanning -> associating
Aug 31 20:35:53 Arch kernel: wlp3s0: associate with [MACofESP] (try 1/3)
Aug 31 20:35:53 Arch kernel: iwlwifi 0000:03:00.0 wlp3s0: disabling VHT as WMM/QoS is not supported by the AP
Aug 31 20:35:53 Arch kernel: iwlwifi 0000:03:00.0 wlp3s0: disabling HT as WMM/QoS is not supported by the AP
Aug 31 20:35:53 Arch kernel: wlp3s0: authenticated
Aug 31 20:35:53 Arch wpa_supplicant[423]: wlp3s0: Trying to associate with [MACofESP] (SSID='pwned' freq=2412 MHz)
Aug 31 20:35:53 Arch kernel: wlp3s0: send auth to [MACofESP] (try 1/3)
Aug 31 20:35:53 Arch kernel: wlp3s0: authenticate with [MACofESP]
Aug 31 20:35:53 Arch wpa_supplicant[423]: wlp3s0: SME: Trying to authenticate with [MACofESP] (SSID='pwned' freq=2412 MHz)
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8840] device (wlp3s0): supplicant interface state: disconnected -> scanning
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8639] Config: added 'psk' value '<hidden>'
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8639] Config: added 'auth_alg' value 'OPEN'
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8639] Config: added 'key_mgmt' value 'WPA-PSK'
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8639] Config: added 'scan_ssid' value '1'
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8639] Config: added 'ssid' value 'pwned'
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8638] device (wlp3s0): Activation: (wifi) connection 'pwned' has security, and secrets exist.  No new secrets needed.
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8635] device (wlp3s0): state change: prepare -> config (reason 'none', internal state 'managed')
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8623] device (wlp3s0): state change: need-auth -> prepare (reason 'none', internal state 'managed')
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8590] device (wlp3s0): supplicant interface state: disabled -> disconnected
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8136] device (wlp3s0): state change: config -> need-auth (reason 'none', internal state 'managed')
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8136] device (wlp3s0): Activation: (wifi) access point 'pwned' has security, but secrets are required.
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8133] device (wlp3s0): state change: prepare -> config (reason 'none', internal state 'managed')
Aug 31 20:35:49 Arch NetworkManager[333]: <info>  [1504204549.8124] device (wlp3s0): supplicant interface state: disconnected -> disabled`

uname -a:
`Linux Arch 4.12.8-2-ARCH #1 SMP PREEMPT Fri Aug 18 14:08:02 UTC 2017 x86_64 GNU/Linux`

tobozo, 2017-08-31:
disable 802.11n

Mr-iX, 2017-09-01:
Unfortunately that does not solve the problem. I still get the same error.

tobozo, 2017-09-01:
It sounds like a NetworkManager problem (the humand kind), not a bug.

Can you paste the content of your WiFi config, device model, and explain how you start the WiFi (manually?) once the deauther is ready ?

Mr-iX, 2017-09-01:
iwconfig:
`wlp3s0    IEEE 802.11  ESSID:"WiFi"  
          Mode:Managed  Frequency:5.26 GHz  Access Point: [MAC]   
          Bit Rate=40.5 Mb/s   Tx-Power=15 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=52/70  Signal level=-58 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:11   Missed beacon:0`

lspci:
`03:00.0 Network controller: Intel Corporation Centrino Ultimate-N 6300 (rev 35)
	Subsystem: Intel Corporation Centrino Ultimate-N 6300 3x3 AGN
	Flags: bus master, fast devsel, latency 0, IRQ 31
	Memory at f7d00000 (64-bit, non-prefetchable) [size=8K]
	Capabilities: [c8] Power Management version 3
	Capabilities: [d0] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [e0] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [140] Device Serial Number [MAC]
	Kernel driver in use: iwlwifi
	Kernel modules: iwlwifi`

I'm trying to connect using the gnome ui.
When I enter the password for the deauther wifi it takes a while, but then it throws me back to the password entry.

tobozo, 2017-09-01:
One thing you can try is to setup your Wireless as "ad hoc" or "infrastructure" and make it use the 2.4Ghz band in the gnome ui.

Mr-iX, 2017-09-02:
The strange thing is, that every WiFi Network works, except the deauther WiFi.

tobozo, 2017-09-02:
Other things to try: use another WiFi dongle, read serial debug from deauther

Also if you have another ESP8266, confirm that you can connect to it from linux.

Mr-iX, 2017-09-02:
Now I have tried it with a Edimax WiFi dongle and it works!

I have already tried multiple ESP8266, but with everyone I have the same problem.
So I think there is a problem with Intel Network Cards and the ESP8266.
The bad thing is, that I cannot ask in the Arch Forum:

> I closed the thread because it seeks advice on a device that is, for
almost every use case, illegal.  I would venture that the problem is
some quirk with the ESP-Deauther and not an issue with NetworkManager.
Regardless, the Arch Linux forums are not going to be used as a forum to
promote the use of a device intended to maliciously interfere with
lawful networks that the script kiddie using the device has no right to
attack.  Perhaps you do represent the 0.1% of the people who have a
legitimate use; I shant debate that.  99.9% of the readers won't be.
I suggest you drop the subject.
Arch Linux Forum Administrator.

Has the precompiled deauther bin file also a serial output? I started the serial monitor in the Arduino IDE with 115200, but there is no output.

tobozo, 2017-09-02:
Yep that's the first suggested solution, maybe some investigation will hint you at the right driver name, the [example](https://github.com/spacehuhn/esp8266_deauther/issues/581#issuecomment-326438329) I pasted you earlier is only using the legacy driver name, and your mileage may vary.

`ls /etc/modprobe.d/iw*` will confirm that, usually it returns only one result

There is serial output by default, you can even send a [reset](https://github.com/spacehuhn/esp8266_deauther/blob/34b5f08f75e560e036ad58626586073e18c9f39b/esp8266_deauther/esp8266_deauther.ino#L618) order. It just won't print anything if you do nothing.

Mr-iX, 2017-09-02:
There is no file in the modprobe folder, because I enabled  802.11n again.
But your suggested method, disabled the 802.11n successfully.

Now serial output is working, but there is no output when I'm trying to connect to the wifi.
I think we can close the issue, because I think that there won't be a solution for that problem.

closed 2017-09-02 by tobozo