SpacehuhnTech/esp8266_deauther #582: Uploading bin files on a mac (Using esptool-gui)

coolneb, 2017-08-31:
Hey! So I have a mac, I tried to upload the bin files using esptool-gui. But this is what I got
https://gyazo.com/0cd566b75b99a7545be7632204b11391

I have also tried esptool-master and still will not work. It would be awesome if you helped me out, Have a great day!

Mr-iX, 2017-09-01:
When you have a NodeMCU, you have to install this serial driver:
https://github.com/nodemcu/nodemcu-devkit/tree/master/Drivers

coolneb, 2017-09-01:
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.1
Traceback (most recent call last):
  File "./esptool.py", line 2524, in <module>
    _main()
  File "./esptool.py", line 2517, in _main
    main()
  File "./esptool.py", line 2240, in main
    esp = ESPLoader.detect_chip(args.port, initial_baud, args.before)
  File "./esptool.py", line 203, in detect_chip
    detect_port = ESPLoader(port, baud)
  File "./esptool.py", line 177, in __init__
    self._port = serial.serial_for_url(port)
  File "/Library/Python/2.7/site-packages/serial/__init__.py", line 78, in serial_for_url
    instance.open()
  File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 289, in open
    self.fd = os.open(self.portstr, os.O_RDWR|os.O_NOCTTY|os.O_NONBLOCK)
OSError: [Errno 16] Resource busy: '/dev/cu.SLAB_USBtoUART'

coolneb, 2017-09-01:
python ./esptool.py --port=/dev/cu.SLAB_USBtoUART  write_flash  -fm=dio -fs=32m 0x00000 /Users/nebster/Downloads/esptool-master/v3525235FIRMWARE.bin

coolneb, 2017-09-01:
That helped so much. Thank you so much man.

closed 2017-09-01 by tobozo