SpacehuhnTech/esp8266_deauther #585: Attacks don't works

C4rl0r0ss1, 2017-09-03:
Hi, can anybody help me, please?All Attacks dont works for me: still 0 pkts.

FietsenRijder, 2017-09-03:
You should reset it.

C4rl0r0ss1, 2017-09-03:
Thank you for your reply.Sorry but how to reset?there's an rst button on the board is that the way?

C4rl0r0ss1, 2017-09-03:
Tried to reset the board with arduino  serial monitor but nothing changes.All attacks still 0pkts/s

FietsenRijder, 2017-09-03:
Do you have the latest version 1.6 installed?  The reset button on the board doesn't work.

C4rl0r0ss1, 2017-09-03:
Yes sir, latest sperimental version installed, i had reset the board but nothing new happend.Same problem all attacks do not works 0pkts
/s...i also tried master branch...same problem.

spacehuhn, 2017-09-03:
Please follow the installation steps in the [readme](https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md) and reflash it.

C4rl0r0ss1, 2017-09-03:
I allready did. I' ll try again.what do you suggest experimental or master?

C4rl0r0ss1, 2017-09-03:
and what about esp8266 module on arduino the readme told about to install 2.0 version but i think is now obsolete...

jLynx, 2017-09-03:
Its not obsolete as this project still uses it

C4rl0r0ss1, 2017-09-04:
Thank u very much.I reflashed the board with the 2.0 esp8266 module.first time all works good, secon time i powered the board start to on/off and as i can see the power led is on and every about 3 sec. turn off and then on again so is impossible for me to connect to the ap cause disconnect me.what can i do?

tobozo, 2017-09-04:
boot loop can be caused by selected an invalid flash size from the board menu, make sure you have selected the correct value, or test them one by one until it works.

also the serial output can be more informative than the led output for such matter

C4rl0r0ss1, 2017-09-11:
deauth attack still won t work for me...on serial monitor i can see 
running Deauth attack... done
getting attacks JSON (249) done
But on my smartphone app i can see firing only 10 pks/s and don t work.
Can anibody also help me with the oled display connections?
http://imagizer.imageshack.us/a/img923/1889/7Ux8uG.jpg
Thank u

tobozo, 2017-09-11:
Your OLED seems well connected, did you uncomment the [USE_DIPSLAY](https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/esp8266_deauther.ino#L18) and [specify your model](https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/esp8266_deauther.ino#L32) (SSD1306 is selected but you can use SH1106 instead)?

10pk/s means it's working, try selecting more than one device on the APScan page, there's an option for that on the settings pages (multiple ssid).

If you con't see the results of the deauthing on the target device (usually a device in a re-connect loop), check that the access point isn't using 802.11n or 5GHz band, those are not affected by the attack although the scan will see the devices.

C4rl0r0ss1, 2017-09-11:
Yes sir i did. I tried now to change to sh1106 display but still won t on

tobozo, 2017-09-11:
- Are you positive this OLED is working fine with this ESP dev module ?
- Can you post a pic showing the back of your display ? We need to be able to see what is the chip select (e.g. usually 0x78) and make sure of its value.
- Can you try to add this at the beginning of the setup() function:
    `Wire.begin(D1, D2);` (don't forget to remove it if the test doesn't work)
    or this (may not work): `Wire.begin(SDA, SCL)` 
- Check the pinout of your ESP dev module, maybe SDA/SCL have been assigned to different pins

C4rl0r0ss1, 2017-09-11:
Here my display. http://imageshack.com/a/img924/8977/zqhmw7.jpg ..i know i m retardet but can you please tell me in detail where to put that line in the setup function?

tobozo, 2017-09-11:
[we're all retarded and living kick-ass lives](https://youtu.be/qkPGlVqqEP0?t=41) anyway :)

Find the setup() function in esp8266_deauther.ino and put the code as follows:

```
void setup() {
  Wire.begin(SDA, SCL);
  (...)
```

The two arguments are forcing the I2C connection to the specified pin numbers, you may try different values ( sometimes I2C is on D6/D5.).

If you selected "Generic ESP82666 module" in the board list, you will probably get an error like D1 and D2 aren't declared. Just remove the "D" and keep the numerical value:

   `Wire.begin(2, 1)`

Now if you still can't it to work, you may want to try the [I2C scanner](https://gist.github.com/tfeldmann/5411375) first until your OLED is correctly detected (should say "I2C device found at address 0x78").

C4rl0r0ss1, 2017-09-11:
LoL i tried I2c scanner but after the charging is complete if i open serial monitor nothing appear and the board looks off

tobozo, 2017-09-11:
Flash size => 1M (256K SPIFFS) ?

C4rl0r0ss1, 2017-09-11:
Yes no output in the serial monitor and the board still off...

tobozo, 2017-09-11:
`Serial.begin(9600);` either you change that value to 115200 in your I2C scanner sketch, or you select 9600 in your serial monitor.

tobozo, 2018-03-24:
fixed in the next version, closing this

closed 2018-03-24 by tobozo