SpacehuhnTech/esp8266_deauther #586: response error attackStart.json

Eliyanow, 2017-09-04:
**How to solve the problem**

**Serial monitor**
`starting AP scan...
MAC - Ch - RSSI - Encrypt. - SSID - Hidden
Scan results: 9
0 - 6c:aa:b3:29:5b:30 - 4 - -64 - none - AS4 Wireless - 0
1 - 6c:aa:b3:69:5b:30 - 4 - -64 - ? - AS4 Wireless (802.1x) - 0
2 - 6c:aa:b3:a9:5b:30 - 4 - -65 - WPA2 - Alice Staff - 0
3 - 6c:aa:b3:29:2e:f0 - 5 - -80 - none - AK4 Wireless - 0
4 - 6c:aa:b3:69:2e:f0 - 5 - -80 - ? - AS4 Wireless (802.1x) - 0
5 - 6c:aa:b3:a3:2e:f0 - 5 - -80 - WPA2 - House Staff - 0
6 - 6c:aa:b3:21:44:70 - 8 - -79 - none - AS4 Wireless - 0
7 - 6c:aa:b3:61:44:70 - 8 - -80 - ? - AS4 Wireless (802.1x) - 0
8 - 6c:aa:b3:a1:44:70 - 8 - -80 - WPA2 - House Staff - 0
scan done
sending AP scan result JSON .........done
`

tobozo, 2017-09-04:
This is not a problem, it's the intended behaviour.

The ClientScan will temporarily disconnect you from the deauther, one of the side effects is the error response from ClientScan.json.

This is written in big red letters on the stations.html page.

Just wait long enough for the client scan to complete before reconnecting your mobile to the deauther.

closed 2017-09-04 by tobozo

Eliyanow, 2017-09-05:
# How ? 

Still the same problem