SpacehuhnTech/esp8266_deauther #587: deauther 1.6 (by dstike) restarts when scanning over 30ssid's

teon, 2017-09-06:
When accessing places where there are over 30 SSIDs, after 2 seconds from starting "Scaning" deauther just restarts...

I have:

https://www.tindie.com/products/lspoplove/wifi-deauther-oled-v2-include-case-and-antenna/

with deauther 1.6

jLynx, 2017-09-06:
Are you on battery only when this happens, or are you plugged in with the USB as well?

tobozo, 2017-09-06:
If you can recompile the sketch, you can set the [maxAPScanResults](https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/APScan.h#L4) to a lower value (i.e 32).

Your scan results will be trimmed but you won't have the reboot problem.

teon, 2017-09-06:
@jLynx on battery. I can check out on cable as well.
@tabozo thanks for the tip, but as I understand it's a workaround?

spacehuhn, 2017-09-06:
It might be a workaground. We're trying to figure out how to fix this bug https://github.com/spacehuhn/esp8266_deauther/issues/579

tobozo, 2017-09-06:
@teon temporary workaround indeed
I'm currently [searching](https://github.com/esp8266/Arduino/issues?utf8=%E2%9C%93&=milestone%3A2.0.0%20) for a possible related bug,

jLynx, 2017-09-06:
@tobozo can you try again with it plugged into power and let me know the result?

tobozo, 2017-09-06:
@jLynx it does it with and without power. Also reproduced on a Wemos D1 Mini Pro, ( `maxAPScanResults 80` and no display) so it's not specific to the OLED build..

```
Exception (29):
epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd80 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff20:  40282490 00000000 0000000d 4020d59c  
3fffff30:  402315f7 3ffef64c 3fff5f48 40231602  
3fffff40:  00000000 402313eb 3ffef664 3ffe9c60  
3fffff50:  021c0fd0 3ffef664 3ffe9c60 021c0fd0  
3fffff60:  4023187c 3ffec660 000000f5 60000600  
3fffff70:  4021bdbd 3ffe9c60 3ffe9ffc 4021bdbd  
3fffff80:  4021be02 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe9c78 3fffdab0 00000000 3fffdcc0  
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```

```
0x40282490: chip_v6_unset_chanfreq at ?? line ?
0x4020d59c: ESP8266WiFiClass::_scanDone(void*, int) at /home/tobozo/.arduino15/packages/esp8266/hardware/esp8266/2.0.0/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp line 952
0x402315f7: scan_cancel at ?? line ?
0x40231602: scan_cancel at ?? line ?
0x402313eb: scan_pm_channel_op_cb at ?? line ?
0x4023187c: scan_clear_channles at ?? line ?
0x4021bdbd: ets_timer_handler_isr at ?? line ?
0x4021bdbd: ets_timer_handler_isr at ?? line ?
0x4021be02: ets_timer_handler_isr at ?? line ?
```

teon, 2017-09-07:
@jLynx both battery and DC/usb restarts

jLynx, 2017-09-07:
Thanks guys, seems like it is a code issue then rather than a hardware issue. Will look into it

tobozo, 2017-09-07:
@jLynx could it be caused by bad timing (e.g. ClientScan::packetSniffer interfering with WiFi.scanNetworks) ?

jLynx, 2017-09-12:
Potentially, @spacehuhn and I need to look into it

spacehuhn, 2017-09-12:
@jLynx probably same bug as https://github.com/spacehuhn/esp8266_deauther/issues/579

tobozo, 2017-09-14:
After testing @dwillmore's idea I found out disabling webserver on my OLED build fixed the issue.
But it could also be my area is less busy than usual so this will need confirmation.

@teon would you mind testing this [branch](https://github.com/spacehuhn/esp8266_deauther/tree/serverless) on your build (it disables the web server) to see if it still crashes in busy areas ?

tobozo, 2018-03-24:
fixed in the next version, closing this

closed 2018-03-24 by tobozo