SpacehuhnTech/esp8266_deauther #590: Simplified Chinese Version

jokin1999, 2017-09-08:
I made a version in Chinese.
[Github](https://github.com/jokin1999/esp8266_deauther_chinese)
By the way,you can build something to support muiltpie languages and I can provide the Chinese version.

tobozo, 2017-09-08:
Hey, thanks for your effort.

While this is very helpful for the current build chances are it may break on next release.

@spacehuhn would it make sense to maintain i10n files where all the Web UI terms are centralized at least in one default language? That would enable the possibility to select a language from the web interface and load complete or partial translation sets.
It only needs migrating the HTML/JavaScript to l10n and bootstrapping the default language that will become the template for all further languages.

jokin1999, 2017-10-09:
I'am waiting for the update to support the language switcher.After that I can merge something which is able to this to support Chinese to this project.
By the way,in Chinese,there are too many useless things sent to my email so I may not get the update notice by github.And recently,my cause is a little heavy.But I will check it as possible as I can.

tobozo, 2017-10-12:
Language switcher has been implemented :package: but I has not been merged into the testing branch (yet).

Can you please test the [spiffs](https://github.com/spacehuhn/esp8266_deauther/tree/spiffs) branch.
The install procedure is the same (load the sketch and compile) but with two extra last steps.

In order to add Simplified Chinese to this build you will need to create a "zh-CN.json" file in the [l10n](https://github.com/spacehuhn/esp8266_deauther/tree/spiffs/esp8266_deauther/data/l10n) folder.

You may then copy the content of the [en.json](https://github.com/spacehuhn/esp8266_deauther/blob/spiffs/esp8266_deauther/data/l10n/en.json) into your file so it is ready to receive translations.
:warning:  please make sure you only modify the JSON *values* (don't translate the property names!).

When ready, you can [check that the JSON is valid](https://jsonlint.com/) and hit the Tools/ESP8266 Sketch Data Upload button in your Arduino IDE.

If everything worked fine, you may access the language switcher from the settings page.

jokin1999, 2017-10-13:
I've just submitted a merge request,but this form is not friendly to SimplifiedChinese grammer,may be we can try to change the form into complete sentences.
The version i just submitted(& merged) was weird somewhere in Chinese,but it was easy accessible to use for Chinese.

tobozo, 2017-10-13:
Yep, some sentences are split when a <a href> link highlights a word. 

`{p1}<a href="/blah">{a1}</a>`

I did not want to put HTML in the JSON as it is not the role of a translation to carry structure information, but could this solve the problem?

`{p1}<a href="/blah">{a1}</a>{p2}`

If every split text is prefixed *and* suffixed by a localisable string, you may use {p1} or {p2} only, or both, depending on the grammar. This would also solve direction problems for Hebrew or Arabic.

Keep in mind that when a new {property} is added to the HTML/JSON, all language files must be updated to reflect the changes, so the sooner it happens is the best.

jokin1999, 2017-10-13:
Yep,thanks for your efforts.

tobozo, 2017-10-14:
You may now safely delete the `spiffs` branch and chackout/pull the `testing` branch.

Do you want to implement the grammar fixtures? 

All you need to do is create a {p1-suffix} or {p1-prefix} where needed in the html templates and add the new properties in the JSON l10n files. 
You can assign an empty value to the other languages so the translators can leave as is or fill at will.

jokin1999, 2017-10-15:
The unmatched things at line 24 called "Name for".Should I add "Name for-prefix" and "Name for-suffix" after or before it?

haoqingtianxia, 2017-11-14:
多谢了，我翻译的时候正好有一些不懂，还有一些不知道该怎么翻译，现在全懂了[image]

jokin1999, 2017-11-14:
有些翻译还是有点问题的，专业术语不太好翻译，最新的翻译在l10n文件夹里。

haoqingtianxia, 2017-11-17:
[image]
我的彩色版完成了，请教一个问题：自己翻译的web_server怎么才上传进开发板里，我是新人上传了好几次还是原版的

tobozo, 2018-03-24:
please check the next release (anytime soon) and open a new issue for any matter regarding internationalization and localization.

closing this as related to the soon-to-be old version

closed 2018-03-24 by tobozo