SpacehuhnTech/esp8266_deauther #591: Reboot loop using v1.6 binaries

dwillmore, 2017-09-08:
As a quick test, I flashed the v1.6 binary to a NodeMCU clone board.  It runs several other example sketches fine--wifi ap scan, etc.  So, I have some confidence that the board is fine.

Serial log:
`Settings:
SSID: pwned
SSID length: 5
SSID hidden: 0
password: deauther
password length: 8
channel: 1
Default MAC AP: 5e:cf:7f:19:20:86
Saved MAC AP: 5e:cf:7f:19:20:86
MAC AP random: 0
Scan hidden APs: 1
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 15
use built-in LED: 1
channel hopping: 0
multiple APs: 0
multiple Attacks: 0
mac change interval: 4
1s beacon interval: 0
LED Pin: 2
settings saved
Settings:
SSID: pwned
SSID length: 5
SSID hidden: 0
password: deauther
password length: 8
channel: 1
Default MAC AP: 5e:cf:7f:19:20:86
Saved MAC AP: 5e:cf:7f:19:20:86
MAC AP random: 0
Scan hidden APs: 1
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 15
use built-in LED: 1
channel hopping: 0
multiple APs: 0
multiple Attacks: 0
mac change interval: 4
1s beacon interval: 0
LED Pin: 2
Trying to sync the MAC addr with settings
Could not sync because settings are not loaded !
nameList cleared
Attack LED : OFF

Starting WiFi AP:

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff5410 end: 3fff5850 offset: 01a0

>>>stack>>>
3fff55b0:  3fff8978 00000001 00000001 ffffff80  
3fff55c0:  40230b56 40223ff4 00000018 3fff8448  
3fff55d0:  00000001 3fff8910 3fff8910 40230d3c  
3fff55e0:  3fff8910 3fff8448 3fff68c8 00000001  
3fff55f0:  3ffef604 3ffef418 61767300 3fff68c8  
3fff5600:  40281958 00000064 197fcf5e fffeffff  
3fff5610:  00000018 0000ffff 00000000 00000001  
3fff5620:  3fff5638 4022d2f0 3ffef394 3fff8448  
3fff5630:  197fcf5e 3fff8620 00000000 00000000  
3fff5640:  00000000 00000000 00000000 00000000  
3fff5650:  00000000 00000000 00000000 00000000  
3fff5660:  00000000 00000000 4022dfe7 3fff8448  
3fff5670:  3fff45f8 3fff8448 3fff8448 4022e02f  
3fff5680:  00000000 3ffef418 3ffea2bc 00000000  
3fff5690:  4021ce29 00000003 00000003 3fff68c8  
3fff56a0:  4021cf41 00000003 00000003 3fff6bac  
3fff56b0:  00000001 4021cf9a 00000003 3fff68c8  
3fff56c0:  4020d660 3fffc6fc 00000001 3fff4758  
3fff56d0:  00000020 3fff68e8 3fff45f8 4020d769  
3fff56e0:  3fff5730 00000001 3fff577c 40212baa  
3fff56f0:  3fff6aa0 0000000a 3fff4758 40210681  
3fff5700:  3fff4758 00000001 3ffe9545 4021212c  
3fff5710:  3fff8478 4010683a 3fff57a0 00000002  
3fff5720:  00000006 0000000a 3fff4758 40212b52  
3fff5730:  3fff8458 4010683a 3fff57ec 00000001  
3fff5740:  00000000 0000000a 3fff4758 40210681  
3fff5750:  3fff4758 00000001 3ffe9545 4021212c  
3fff5760:  3fff4758 0000000f 3ffe899b 3fff481c  
3fff5770:  00000000 00000010 3fff4758 40210a81  
3fff5780:  3fff4758 00000001 3ffe9545 4021212c  
3fff5790:  3fff4758 00000011 3ffe8f2d 3fff481c  
3fff57a0:  3fff4758 00000012 3fff4758 40210a81  
3fff57b0:  4021cdc2 3fff45f8 3fff4758 3fff45f8  
3fff57c0:  00000000 3fff06d0 3fff4758 40212200  
3fff57d0:  00000001 401011f3 3fff4758 3fff481c  
3fff57e0:  3fff45f8 3fff4758 3fff06d0 4020cc5f  
3fff57f0:  402012e6 000001f4 000001f4 402012db  
3fff5800:  3fff4758 3fff06d0 3fff072c 4020cf3b  
3fff5810:  00000000 00000000 00000000 00000000  
3fff5820:  00000000 00000000 00000000 3fff481c  
3fff5830:  3fffdc20 00000000 3fff4815 40210cd6  
3fff5840:  00000000 00000000 3fff4830 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
Settings:
SSID: pwned
SSID length: 5
SSID hidden: 0
password: deauther
password length: 8
channel: 1
Default MAC AP: 5e:cf:7f:19:20:86
Saved MAC AP: 5e:cf:7f:19:20:86
MAC AP random: 0
Scan hidden APs: 1
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 15
use built-in LED: 1
channel hopping: 0
multiple APs: 0
multiple Attacks: 0
mac change interval: 4
1s beacon interval: 0
LED Pin: 2
Trying to sync the MAC addr with settings
Synced with saved mac addr : 5e:cf:7f:19:20:86
nameList cleared
Attack LED : OFF

Starting WiFi AP:

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff5410 end: 3fff5850 offset: 01a0

>>>stack>>>
3fff55b0:  3fff8978 00000001 00000001 ffffff80  
3fff55c0:  40230b56 40223ff4 00000018 3fff8448  
3fff55d0:  00000001 3fff8910 3fff8910 40230d3c  
3fff55e0:  3fff8910 3fff8448 3fff68c8 00000001  
3fff55f0:  3ffef604 3ffef418 61767300 3fff68c8  
3fff5600:  40281958 00000064 197fcf5e fffeffff  
3fff5610:  00000018 0000ffff 00000000 00000001  
3fff5620:  3fff5638 4022d2f0 3ffef394 3fff8448  
3fff5630:  197fcf5e 3fff8620 00000000 00000000  
3fff5640:  00000000 00000000 00000000 00000000  
3fff5650:  00000000 00000000 00000000 00000000  
3fff5660:  00000000 00000000 4022dfe7 3fff8448  
3fff5670:  3fff45f8 3fff8448 3fff8448 4022e02f  
3fff5680:  00000000 3ffef418 3ffea2bc 00000000  
3fff5690:  4021ce29 00000003 00000003 3fff68c8  
3fff56a0:  4021cf41 00000003 00000003 00000001  
3fff56b0:  00000001 4021cf9a 00000003 3fff68c8  
3fff56c0:  4020d660 3fff8498 04000102 3fff848a  
3fff56d0:  00000012 3fff68e8 3fff45f8 4020d769  
3fff56e0:  00000002 3fff5724 000000a2 00000015  
3fff56f0:  00000001 0000001c 0001c200 00000000  
3fff5700:  0000ffff 0000ffff 3fff5880 4020387c  
3fff5710:  3fff8488 4010683a 3fff57a0 00000002  
3fff5720:  3fff7ed4 00000000 00000000 40212b52  
3fff5730:  4010684c 3fff57ec 3fff57ec 00000001  
3fff5740:  000bb0fe 0000000a 3fff4758 40210681  
3fff5750:  3fff4758 00000001 3ffe9545 4021212c  
3fff5760:  3fff4758 0000000f 3ffe899b 3fff481c  
3fff5770:  00000000 00000010 3fff4758 40210a81  
3fff5780:  3fff4758 00000001 3ffe9545 4021212c  
3fff5790:  3fff4758 00000011 3ffe8f2d 3fff481c  
3fff57a0:  3fff4758 00000012 3fff4758 40210a81  
3fff57b0:  4021cdc2 3fff45f8 3fff4758 3fff45f8  
3fff57c0:  00000000 3fff06d0 3fff4758 40212200  
3fff57d0:  00000001 401011f3 3fff4758 3fff481c  
3fff57e0:  3fff45f8 3fff4758 3fff06d0 4020cc5f  
3fff57f0:  402012e6 000001f4 000001f4 402012db  
3fff5800:  3fff4758 3fff06d0 3fff072c 4020cf3b  
3fff5810:  00000000 00000000 00000000 00000000  
3fff5820:  00000000 00000000 00000000 3fff481c  
3fff5830:  3fffdc20 00000000 3fff4815 40210cd6  
3fff5840:  00000000 00000000 3fff4830 40100114  
<<<stack<<<`

After a normal startup, it creates the settings and restarts, then it crashes.  After that, it just crashes after starting.  Any suggestions?

tobozo, 2017-09-08:
Can you reflash it after selecting `1M (256K SPIFFS)` from the Tools/Flash Size menu.
You may have to manually choose `Generic ESP8266 Dev Module` in the Board list to see that.

dwillmore, 2017-09-08:
Thanks for the reply.  I'm just using espytool.py to flash it, there aren't any such settings as those go into the packaging of the binary file--which has already been done for the binaries.

Or am I misunderstanding something?

dwillmore, 2017-09-08:
Should it matter that I have a board with a 4MB (16Mb) FLASH?  The onlything at the top of memory are the AP settings and the calibration info, IIRC--which doesn't need rewritten when you flash a program.

tobozo, 2017-09-08:
The [Settings.h](https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/Settings.h) is full of hardcoded addresses, I'm not sure how this behaves with different layouts though.
I just know that if I flash my modules with another FlashSize setting from Arduino IDE, it goes into boot loop or just doesn't work.

dwillmore, 2017-09-08:
I'll look into setting up an Arudino boot environment and building from souce.  I was hoping to avoid that just to try deauther out.  Oh, well, time to start digging.

Thanks for the help.

dwillmore, 2017-09-08:
And that works.  Thanks for your help.  Setting up the build environment was pretty simple.

closed 2017-09-08 by tobozo