SpacehuhnTech/esp8266_deauther #593: error loading attackstart.json

talal2, 2017-09-08:
Hi 
I Got connected with yellow symbole in my wifi, but i still entered to my esp board, with IP, but when i launch the attack, i got this error: error loading attackstart.json
I Did the sme steps in this video: https://www.youtube.com/watch?v=1cpVJTcHK7U

marek26340, 2017-09-09:
Try enabling the "LED while attacking" option in the settings. If it will light up on the ESP when you start attack, that means the ESP has changed it's WiFi channel to the channel of the WiFi you want to attack. Simply reconnect WiFi, go to 192.168.4.1 and everything should be fine. If it is not lighting up, then please copy the output from the serial monitor in Arduino IDE and paste it here or on pastebin, whatever, so we can get an idea of what's wrong.

tobozo, 2017-09-21:
Launching the attack will disconnect you (there's a big message in red letters telling you this).
Feel free to reopen this if you think it's a bug.

closed 2017-09-21 by tobozo