SpacehuhnTech/esp8266_deauther #594: The website doesn't show Wifi access points

Valentijn12345, 2017-09-10:
When i press the "scan" button, it says: scanning...
and in a few seconds it appears....

please help me

tobozo, 2017-09-10:
Please specify your hardware and also the software version.
Also give more details on the symptoms because '...' only leaves space to imagination.

Did Cthulu appear ?

jLynx, 2017-09-12:
So it does or doesn't show them? In the title you say they dont show but in the body of you message you say they do appear. Please give more detail

rbomze, 2017-09-20:
Yeah... i found the reason for this problem. When there is an WPA2-Enterprise encryption enabled the integer in the JSON response turns to '255', therefor calculated Response-length of the JSON response is too short, therefor the browser reads an incomplete and malformed JSON response.
Fix (in APScan.cpp):
    //_size += 61;   //comment out, as value is not only single digit 
    _size += 60       //
    _size += String(encryption[i]).length();      //calculate correct string length

tobozo, 2017-09-21:
@rbomze it is not clear whether `_size += String(encryption[i]).length(); ` needs to be inserted or replaces another statement. Would you be kind enough to make a PR on the testing branch for that ?

tobozo, 2018-03-24:
fixed in the next version, closing this

closed 2018-03-24 by tobozo