SpacehuhnTech/esp8266_deauther #602: i get (internet may not be available)

101tuck, 2017-09-13:
i try to go to the ip and it will not load can someone help meh plz

jLynx, 2017-09-13:
So what IP are you trying to load, what is the SSID name...etc. You need to give us more information than that if you want help

101tuck, 2017-09-13:
i try to load pawnd i put the pass in and i go to  192.168.4.1 put it says there is no internet

tobozo, 2017-09-13:
Please give us more information, we can help but we won't assist.

The "more info" you can provide us with *details* on:

- what ESP are you using ?
- did you flash it yourself ? if so, what version of Arduino, what Operating system are you using ?
- what device are you using when you connect to 192.168.4.1 ? what browser ?

please edit your first comment and add those details

101tuck, 2017-09-13:
1.HiLetgo New Version NodeMCU LUA WiFi Internet ESP8266 Development
2.i did flash it myself 64 bit operating system
3.useing my phone and i try it on the computer and useing chrome

tobozo, 2017-09-13:
possible causes:
- flashing failed => you need to tweak some settings in Arduino before recompiling and flashing
- power problem => ESP8266 is spec'd at 250mA, make sure your power supply can provide that
- boot problem => plug the ESP to the PC and open your serial monitor (115200bauds), push the reset button on the ESP, what does it say in the serial monitor ?

101tuck, 2017-09-13:
>flashing failed=>what setting would i mess with 
>power problem=the power supply dose provide that
>boot problem=open serial monitor????
(Kind of new to this)

101tuck, 2017-09-13:
is there any way to reset the bored so it is like you just took it out of the package ??

sean-the-random, 2017-09-13:
Yes go to the update 1.6 and ree inject it and its new agen...

tobozo, 2017-09-13:
Chances are the binary you used to flash your ESP is pre compiled for a different model (i.e. not the same amount of flash memory), you might be more lucky by compiling from Arduino IDE.
This is where you'll find the serial monitor.
If you're new to this maybe put this project on wait and get familiar with Arduino IDE by doing trivial stuff first (blink the led, do a wifi scan).

101tuck, 2017-09-13:
ok i got working i can load the 192.168.4.1 but i get error loading attackstart.json?num=0

101tuck, 2017-09-13:
truns out i was not getting the right amount of power

closed 2017-09-13 by tobozo