SpacehuhnTech/esp8266_deauther #603: deauther esp

thegitter12345, 2017-09-13:
I need a little help. I turned the board on and it stayed on for an hour then turned off. I could not get it back on for a half hour. when it did come back on the credentials changed to hidden network from my ssid. Now I cant log on. Before I get a refund I want to see if I can't fix this. This is the model below:

https://www.tindie.com/products/lspoplove/wifi-deauther-oled-v2-include-case-and-antenna/

spacehuhn, 2017-09-13:
Maybe have a look at the FAQ: https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ

tobozo, 2017-09-29:
@thegitter12345 did reading the FAQ solve your issue? If so, can we close this issue?

tobozo, 2017-10-16:
closing this, feel free to reopen when new details come up

closed 2017-10-16 by tobozo