SpacehuhnTech/esp8266_deauther #604: Feature proposal: option to remove web server

dwillmore, 2017-09-14:
Hello, all.  I've been working with esp8266-deauther with an OLED display.  In that use case, I have no need for the web UI.  Would a patch to disable that portion of the code based on a config option be welcomed?

tobozo, 2017-09-14:
You can disable the WebUI from the menu by setting the WiFi off, is that what you mean ?

dwillmore, 2017-09-14:
I saw that and wondered what it did--for me all it does is drop the power consumption below the threshold that my USB supply thinks the attached device is done and powers off.  So, that's been a convenient 'power device off'.

But, no, I was thinking of a #define that would disable inclusing of the web UI at build time.

tobozo, 2017-09-14:
Your suggestion is very helpful indeed.

I just did a quick test using something like `#ifdef USE_WEBSERVER` on my modified local version and found out this is a better workaround for an annoying bug we're trying to squash, see #208 and #587 

Despite the fact it isn't on the roadmap, I'm quite sure this patch will be welcome.

dwillmore, 2017-09-14:
Oh, how fortunate!  Would you like me to make up a patch or--since you already seem to have done it--would you like to take this up?  Would an option to build without serial support also be allowed?

I'm just looking for ways to use less program memory so that an OLED or similar display would work with a full MAC database on a 1MB board like the esp8285.

tobozo, 2017-09-14:
Yes please submit a PR to the testing branch.

Just ignore the new [branch](https://github.com/spacehuhn/esp8266_deauther/tree/serverless) it's only here as a POC.

dwillmore, 2017-09-14:
For the serial port disabling?

tobozo, 2017-09-14:
I meant for the web server support

Disabling serial would be another feature request I guess, and probably subject to discussion too as it is a bit of an aggressive change towards the default behaviour of the tool.

dwillmore, 2017-09-14:
Understood.  I just thought I'd mention it.  I'll open a different issue for that.  I'll also go check out the testing branch and make the changes to that to remove the web UI.

dwillmore, 2017-10-02:
Yeah, this one will be pretty huge.  We might want to wait until after the next major revision.

tobozo, 2018-03-24:
implemented in the next version, closing this

closed 2018-03-24 by tobozo

tobozo, 2018-03-31:
@dwillmore have you tried with the esp8285 since the new version is out ?

dwillmore, 2018-03-31:
@tobozo Nope, sorry, I was at a conference last Saturday when you were closing all those issues and releasing 2.0.  Great work!  I've been swampled since them, but I'll get around to checking it out.  I'll check out the 'disable serial' option as well.  Then I'll rebase my Quadrature Encoder UI code for 2.0 if that's all right?