SpacehuhnTech/esp8266_deauther #605: Feature Proposal: Allow disabling of the serial interface

dwillmore, 2017-09-14:
Like my previous issue to disable the code in esp8266_deauther which implements the web UI, this issue is to propose to provide a #define to disable the serial output.  This has fewer benefits than the web UI disabling, but isn't needed for portable uses or for those using just the web UI.  But serial port support does drag in a lot of code.

This would complete the UI enabling/disabling features that I have in mind.  We would end with three separate #define statements.  One for the web UI, one for the serial output (can't really call it a UI), and one for the OLED UI.  If none of them are selected a #error would be triggered to let the user know that they're basically asking for a null set of functionality.

tobozo, 2017-09-29:
Serial.print() statements using quoted strings are eating a lot of memory, does it make a big difference if only those statements are enabled/disabled by #define ?
This way the deauther is still listening to serial messages such as reset.

spacehuhn, 2017-09-29:
I don't think it will make any noticable difference. I mean we're talking about an 80-160MHz WiFi SoC.
You can change it to make it over #define, but then you have to `#define debug` in every single file and every little serial log will create atleast 2 extra lines, which doesn't improve the readability.

exploitagency, 2017-09-29:
Is there enough flash left over to use the F() macro with your Serial.print() statements?  I don't know if the older ESP release this project uses supports it but if not there are other ways to do the same thing, this way is just one of the easiest.  I have never looked at this project in depth but I am watching it and just saw the issue.

https://playground.arduino.cc/Learning/Memory

Examples

Serial.println(F("text"));

Lazy way to combine F() with a variable:
String()+F("text")+variable

Hope this was useful

dwillmore, 2017-10-02:
@exploitagency, I don't think that'll help much on the esp8266 as everything is in flash anyway.  That macro is left over from the Atmel Arduino days.

tobozo, 2017-10-02:
@dwillmore now I'm confused about [this commit](https://github.com/esp8266/Arduino/commit/7450dd39ca93ca849f9198b0151e4c7c747e43ef#diff-a74cc817f4b376e0d5661de099aea3db)

dwillmore, 2017-10-02:
With the AVR 8 bit chips, strings are stored in flash and are coppied to RAM when they are used.  That wastes RAM for constant strings.  So, they came up with this F() wrapper to declare them to stay in flash.  So, once you treat them as a const, they can stay in flash and not need to be copied to RAM and use up space there with a duplicate.

I really don't know that it does on esp8266, but since it's a unified memory space instead of a split memory space, I would think that construct would no longer be necessary.

I'd be very curious what this actually does.  Declaring the strings as const should be enough, I would think.

exploitagency, 2017-10-03:
I have actually never verified how it plays out on the chip itself but this is what the Arduino IDE says, which has been wrong before.

```
//Test Case without F() Macro

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  Serial.println("test a long string to print to serial console here");
  // put your main code here, to run repeatedly:

}
```

Sketch uses 224133 bytes (51%) of program storage space. Maximum is 434160 bytes.
Global variables use __31676__ bytes (38%) of dynamic memory, leaving 50244 bytes for local variables. Maximum is 81920 bytes.

----------------

```
//Test Case with F() Macro

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  Serial.println(F("test a long string to print to serial console here"));
  // put your main code here, to run repeatedly:

}
```

Sketch uses 224213 bytes (51%) of program storage space. Maximum is 434160 bytes.
Global variables use __31628__ bytes (38%) of dynamic memory, leaving 50292 bytes for local variables. Maximum is 81920 bytes.

dwillmore, 2017-10-03:
So, it looks like it may save RAM but does nothing for flash.  That's pretty much what I suspected.  Probably wouldn't hurt to declare all the tables and strings const and static.

exploitagency, 2017-10-03:
Yeah that was my intention, I thought the issue here was to save RAM so I just threw it out there for you.  I have never looked at the code to this project to see the flash requirements or how much is needed for any sort of datalogging.

tobozo, 2018-03-24:
closing this as things changed in the new version

closed 2018-03-24 by tobozo

spacehuhn, 2018-03-24:
you can disable the serial interface in the settings with the 2.0 version but it's not recommended!