SpacehuhnTech/esp8266_deauther #607: fatal error: include/wl_definitions.h: No such file or directory

chamithjay97, 2017-09-15:
before i compiled scripts to my nodemcu, but however not scripts dont compiled... this is the full error..

In file included from D:\death\esp8266_deauther\esp8266_deauther.ino:11:0:

C:\Users\JAY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include/ESP8266WiFi.h:28:38: fatal error: include/wl_definitions.h: No such file or directory

   #include "include/wl_definitions.h"

                                      ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

(error saying that wl_definitions.h is not in directory... but it is in there.. and i downloaded it again and put, but dont work..installed IDE several times... i done all steps in your github... but didnt work yet... can you please help me to fix these...)

tobozo, 2017-09-15:
looks like you didn't follow all the steps in the installation instructions

chamithjay97, 2017-09-16:
i followed the all steps... trust me.. after any update on ide its not compiled... i got all the way this error after that

narcomerk19, 2017-09-16:
try on other computer, with all admin priviledge ☺
I think the problem is in your pc.

tobozo, 2017-09-16:
well read again then

13 Copy ESP8266Wi-Fi.cpp and ESP8266Wi-Fi.h

14 Paste these files here packages > esp8266 > hardware > esp8266 > 2.0.0 > libraries > ESP8266WiFi > src

you mixed step 9 and 14, now you have unnecessary files in your sdk/include folder and the IDE complains about it

chamithjay97, 2017-09-16:
so see this https://ibb.co/igGWjQ

am i wrong? and still getting that problem... :/

chamithjay97, 2017-09-16:
oh my dear sir,, you are a god!!! you exactly right!!! unnecessary files in my sdk... i added that ESP8266Wi-Fi.cpp and ESP8266Wi-Fi.h to sdk thats why i got this problem!!!! oh my god thanks sir!!!! you are a god!!! no words!!!! i love you sir!!!! thank you very much!!!! oh god.... thank you again sir!!!! you are a legend!!!!

closed 2017-09-16 by tobozo