SpacehuhnTech/esp8266_deauther #610: Start scanning APs, the system will crash

rfvcdewsxzaq, 2017-09-16:
The following is the serial port to monitor the data：

ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
Settings:
SSID: ESP8266_Deauther
SSID length: 16
SSID hidden: 0
password: deauther
password length: 8
channel: 1
Default MAC AP: 5e:cf:7f:81:02:ab
Saved MAC AP: 5e:cf:7f:81:02:ab
MAC AP random: 1
Scan hidden APs: 1
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 15
use built-in LED: 1
channel hopping: 0
multiple APs: 1
multiple Attacks: 1
mac change interval: 4
1s beacon interval: 1
LED Pin: 2
Trying to sync the MAC addr with settings
Synced with a random mac addr : 00:1e:63:b1:88:4e
nameList cleared
Attack LED : OFF

Starting WiFi AP:
SSID     : 'ESP8266_Deauther'
Password : 'deauther'
-----------------------------------------------

 generating MACs...done

Starting...

sending AP scan result JSON done
starting AP scan...
MAC - Ch - RSSI - Encrypt. - SSID - Hidden

Exception (29):
epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd80 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff20:  402829e0 00000000 0000000d 4020d57c  
3fffff30:  402315ec 3ffef91c 3fff6218 40231606  
3fffff40:  00000000 402313ef 3ffef934 3ffe9f30  
3fffff50:  035c36fd 3ffef934 3ffe9f30 035c36fd  
3fffff60:  40231880 3ffec2ec 000000f5 60000600  
3fffff70:  4021bdc9 3ffe9f30 3ffea2cc 4021bdc9  
3fffff80:  4021be0e 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe9f40 00000000 3fff8d50 4021ba7f  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

tobozo, 2017-09-16:
same as #587 and #609 

please specify your hardware, are you using the OLED version ?

rfvcdewsxzaq, 2017-09-18:
I do not use the OLED version, how to specify the version

rfvcdewsxzaq, 2017-09-18:
ESP-12-E Series  what should I do
[image]
[image]

rfvcdewsxzaq, 2017-09-18:
I have set maxAPScanResults to a lower value, set from 80 to 30, but still have a crash

tobozo, 2017-09-18:
Speculations: the WebServer crashes the ESP when memory is exhausted. The more SSIDs around you, the faster it crashes. Memory mapping is also weird on some models, probably caused by problems in the sdk 2.0.0, which we can't fix because it is the only SDK that allows deauthing.

Can you try to compile/upload using those settings and paste the serial output ?

- Board: Generic ESP8266
- Flash Size: 1M (256K SPIFFS)

rfvcdewsxzaq, 2017-09-22:
And did not solve the problem

tobozo, 2017-09-29:
how do you power your build ?
Can you afford buttons and OLED? Many issues are related to the Webserver and WebUI, using the OLED and buttons is a way to bypass it, and there's a serverless branch currently being tested too.

rfvcdewsxzaq, 2017-09-30:
No  OLED

tobozo, 2017-09-30:
Please answer all the questions and don't be afraid to leave long answers :-)

How you power your build? Regulated current? What's the voltage/power?
Can you try to cover the antenna (so the signal quality is lower), then see if it still crashes?

Can you afford buttons? Would you be ready to try (modify, compile, upload) a different version of this project?

tobozo, 2018-03-24:
fixed in the next version, closing this

closed 2018-03-24 by tobozo