SpacehuhnTech/esp8266_deauther #619: How could we make wireshark vendor list to data.h

Behooz-Sirouszad, 2017-09-24:
iv seen the wireshark vendor list but icant find any converter or how to convert them like code in data.h?

tobozo, 2017-09-29:
Your request sounds related to @dwillmore's [idea](https://github.com/spacehuhn/esp8266_deauther/issues/604#issuecomment-329590734) to host a full mac database.

rubfi, 2017-10-13:
I've created a small script in order to update the MAC manufacturers list, please, take a look to #638

tobozo, 2017-10-16:
changes merged info the testing branch, the ReadMe has been updated with instructions

closed 2017-10-16 by tobozo

tobozo, 2017-10-22:
@dwillmore @rubfi @Behooz-Sirouszad I've been testing different situations after I updated the vendorlist using @rubfi's python utility.
It works fine on most builds but on DSTIKE it makes the web server crash earlier.

Web server stability comes back if the vendor list is shorter, so I've tested another approach where vendor name is 7-bit encoded in memory, and only uses 6 chars long.

Also I'm not that good at coding python so I've made a PHP script where you can choose the vendorname lenght.

Vendorlist branch (7-bit encoded, vendor len = 6 chars):

```
Sketch uses 497,496 bytes (65%) of program storage space. Maximum is 761,840 bytes.
Global variables use 55,278 bytes (67%) of dynamic memory, leaving 26,642 bytes for local variables. Maximum is 81,920 bytes.
```

Vendorlist branch (7-bit encoded, vendor len = 8 chars):
```
Sketch uses 531,424 bytes (69%) of program storage space. Maximum is 761,840 bytes.
Global variables use 55,342 bytes (67%) of dynamic memory, leaving 26,578 bytes for local variables. Maximum is 81,920 bytes.
```

Testing branch (8-bit encoded, vendor len = 8 chars):

```
Sketch uses 581,840 bytes (76%) of program storage space. Maximum is 761,840 bytes.
Global variables use 57,118 bytes (69%) of dynamic memory, leaving 24,802 bytes for local variables. Maximum is 81,920 bytes.
```